pymmcore-plus/napari-micromanager #22: Contrast Limits - Autoadjust + histogram

ianhi, 2021-07-07:
### Problem

Currently the contrast limits are set by whatever the first image is. This results in behavior like this:
[image]

1st snap was of the random noise of `CY5` with demo cam, second is switching to FITC which has greater intensity which then saturates and the Napari contrast slider don't auto adjust.

### Proposed Solution

1. First pass. Implement something like micromanager `auto adjust contrast limits`
2. Also provide a small histogram (matplotlib based?) as a widget (maybe in the bottom right?) (x-ref https://github.com/napari/napari/pull/675)

tlambert03, 2021-07-07:
yeah, definitely annoying.  I kinda feel like this belongs upstream in napari.  The bones are there (right click on the contrast slider and click "reset" to autoscale).  I think it'd be nice if there was a "keep" autoscale in napari too.  Nonethess, it could be implemented locally as a start

tlambert03, 2022-02-28:
> I think it'd be nice if there was a "keep" autoscale in napari too

napari has this now

closed 2022-02-28 by tlambert03