pymmcore-plus/napari-micromanager #23: MDA Position buttons (Add/Remove/Clear) are non-functional

ianhi, 2021-07-13:
On current `main` the position buttons don't render properly and also don't respond when I click them:

[image]

tlambert03, 2021-07-15:
if you enable the group do they work? (the checkbox next to "Stage Positions")?

ianhi, 2021-07-15:
> (the checkbox next to "Stage Positions")?

:O I never saw that until now! It camouflaged so well. Is there an easy way to change the styling to make that more visible? I truly don't think I would have noticed that without it being pointed out to me.

tlambert03, 2021-07-15:
this is probably an OS-specific issue in the napari stylesheet (and should be fixed in napari).  We don't use [QGroupBox](https://doc.qt.io/qt-5/qgroupbox.html) much over there.  looks like this on mac:

[image]

ianhi, 2021-07-15:
Closing to be dealt with in napari

closed 2021-07-15 by ianhi