pymmcore-plus/napari-micromanager #27: Development tips?

ianhi, 2021-07-27:
Any tips+tricks for developing a napari plugin? In particular is there an easy way to start up napari and immediately load the plugin? Every time I want to test something I am opening napari from the terminal, loading the plugin and then loading the demo config.

tlambert03, 2021-07-27:
> In particular is there an easy way to start up napari and immediately load the plugin?

tlambert03, 2021-07-27:
and here's a script for adding a plugin dock widget to an existing viewer via a script:

```py
import napari

v = napari.Viewer()
dw, main_window = v.window.add_plugin_dock_widget('micromanager')

core = main_window._mmc
core.loadSystemConfiguration('demo')

napari.run()
```

closed 2022-02-28 by tlambert03