pymmcore-plus/napari-micromanager #34: Live mode very laggy on Olympus IX83 on windows

ianhi, 2021-08-13:
Experienced some weird lagginess in live mode today:

https://user-images.githubusercontent.com/10111092/129297071-d0263234-032b-4262-8c31-bd1e50215b92.mp4

It subsequently got much worse, not sure about reproducing as this could also be an issue with pymmcore-plus. I will try to try on a different machine soon, but wanted to open in case anyone else is having

ianhi, 2021-09-01:
I have tried this on a different computer and microscope and experienced the same. So I'm converting this to an issue to discuss ways to improve live mode performance. This is inspired by the complexity in micromanager's live mode which is implemented in this file:
https://github.com/micro-manager/micro-manager/blob/v2.0.0/mmstudio/src/main/java/org/micromanager/internal/SnapLiveManager.java

Among other things they:

1. Use `getLastImage` to grab images
2. call `startContinuousAcquisition(0)` (vs ours with a finite value)
3. adaptively update their image sampling and display rate: https://github.com/micro-manager/micro-manager/blob/4a5d51ea76f89eaa6e74d0d772822701661aa76a/mmstudio/src/main/java/org/micromanager/internal/SnapLiveManager.java#L309-L337

as a start I found that just implementing point 1 results in a significant performance boost but in time it will surely be helpful to take the lessons they learned and more closely mirror their live mode implementation.

tlambert03, 2021-09-01:
very helpful!

tlambert03, 2022-02-28:
can this be closed?

tlambert03, 2023-07-28:
I feel like lots of stuff has been improved here, and this issue is awaiting a response for over a year.  So I'm going to close it.  Please open new issue if there are remaining performance issues with live mode

closed 2023-07-28 by tlambert03