pymmcore-plus/napari-micromanager #42: Generalize `Objective`

ianhi, 2021-09-01:
Currently the device name for the objective is hardcoded as `"Objective"` https://github.com/tlambert03/napari-micromanager/blob/5a8145de33bdd8df7a3d8c4d9f0594fc10a39dd0/micromanager_gui/main_window.py#L266-L269

however this device can be named anything. For example the default micromanager hardware wizard for the Nikon TE-2000 calls it the `nosepiece` which results in napari-micromanager not allowing switching options. 

Two proposed solutions:

1. Require that users create an `Objective`  config group and use that (a la channel)
2. Require that the device be named `Objective` (current state of affairs)
3. Search through all `StateDevices` and match common names (`[O,o]jective`, `[N,n]osepiece`)

fdrgsp, 2021-09-02:
can this #43 (get objectives from an objective group configuration ) be an option?

ianhi, 2021-09-02:
Maybe we should do `3` with a fallback of `1` where we also check for other common names. 

For checking names I would propose starting with this regex:

```python
import re
re.compile("(Nosepiece|Objective|obj)s?", re.IGNORECASE)

```

ianhi, 2021-09-02:
Oh actually it should be `1` with a fallback of `3` so that users can override the devices via a config group

tlambert03, 2021-09-02:
Yep I'm fine with these.  @fdrgsp, you could incorporate the regex checking into #43.  (i.e. instead of `if cfg == "Objectives":` you'd do:

```py
import re
# top of file
OBJ_PTRN = re.compile("(Nosepiece|Objective|obj)s?", re.IGNORECASE)

def _refresh_objective_options(self):
    for cfg in self._mmc.getAvailableConfigGroups():
        if OBJ_PTRN.match(cfg):
            self.objective_comboBox.clear()
            self.objective_comboBox.addItems(self._mmc.getAvailableConfigs(cfg))
            return
```

we could also add a hook in `loadSystemConfiguration` that tries to determine what the "Objective" is going to be using `1` with a fallback of `3` so we don't have to search each time

ianhi, 2021-09-02:
adding the fallback would then be adding a second for loop (only reachable if we didn't already return) that would look like this:

```python
        for device in self._mmc.getLoadedDevicesOfType(pymmcore_plus.DeviceType.StateDevice):
            if OBJ_PTRN.match(device):
                ...
```

tlambert03, 2022-02-28:
can this be closed?

ianhi, 2022-02-28:
I believe so

closed 2022-02-28 by ianhi