pymmcore-plus/napari-micromanager #44: Create a tab for handling arbitrary config groups

ianhi, 2021-09-02:
## Problem

Currently there is no way to modify a config group that is not special cased into the default UI (e.g. Channel). In contrast micromanager doesn't special case any of the config groups and instead devotes a large amount of UI real estate to changing the config groups:
[image]

## Proposed Solution

Either add a new tab to the bottom element (the arrow) or fill in the space (the box) with a config switching widget:
[image]

fdrgsp, 2021-11-30:
@ianhi I have also started working on the group/presets but for the same reason as in #78  I'm waiting to open a PR.

[image]
[image]

ianhi, 2023-02-08:
indeed it is!

closed 2023-02-08 by ianhi