pymmcore-plus/napari-micromanager #78: Generalize (or hide) the autofocus up/down

ianhi, 2021-11-30:
Currently there is a PFS-Offset control add by @fdrgsp in #68 

But this is specific to Nikon scopes, and not every scope will have autofocus. So this element should only be added if one is available (this is what MM does)

Or potentially there should be a dropdown for setting the Focus Device and we should only have a widget for the core focus device. 

Two other issues:
1. The up and down buttons are missing arrows
2. This only added a GUI element, but there doesn't seem to be any code/callbacks connected to it.
[image]

fdrgsp, 2021-11-30:
@ianhi I did already continue developing the GUI in that direction but I wanted to close/merge the PR theta are currently open before opening a new PR (however the "PSF Offset" GUI elements is currently always there and not called only if there is an autofocus device). 

[image]

tlambert03, 2023-07-28:
@ianhi @fdrgsp, please update status and/or close this

fdrgsp, 2023-07-28:
not using this strategy anymore. closing.

closed 2023-07-28 by fdrgsp