pymmcore-plus/napari-micromanager #88: icons do not show up when starting napari with plugin from command line

ptbrown1729, 2022-01-23:
I am running python 3.9.9, napari version 0.4.12, and napari-micromanager installed from github version 0.0.1rc6.dev9+g69c3bed.d20220117 on Windows 10 Pro version 10.0.19041

When I start napari and the napari-micromanager plugin using
`napari -w micromanager`
the icons (e.g. the up/down/right/left arrows for the stage and camera icon on the live button) are not shown. Instead there are blank square placeholders. This may be a more general napari issue, because the napari buttons (e.g. layers buttons, toggle n-display, etc) also do not show any icons. 

On the other hand, if I first launch napari from the command line
`napari`
and then navigate to the plugins menu and select the micromanager plugin, the icons appear as expected

tlambert03, 2022-01-23:
thanks @ptbrown1729, can you tell me what Qt backend version you are on?

tlambert03, 2022-01-23:
actually, more specifically... maybe just give me the output of `conda list`  (or pip list if conda is not involved)

ptbrown1729, 2022-01-23:
Here is the output from `napari.sys_info()`
```
napari: 0.4.12
Platform: Windows-10-10.0.19041-SP0\nPython: 3.9.9 | packaged by conda-forge | (main, Dec 20 2021, 02:36:06) [MSC v.1929 64 bit (AMD64)]
Qt: 5.12.9\nPyQt5: 5.12.3
NumPy: 1.21.2\nSciPy: 1.7.3
Dask: 2021.12.0
VisPy: 0.9.4
OpenGL:\n  - GL version:  4.3.0 - Build 20.19.15.4531
  - MAX_TEXTURE_SIZE: 16384
  - Screens:
  -  - screen 1: resolution 1920x1080, scale 1.0
  - Plugins:
  -   - console: 0.0.4
  -   - micromanager: 0.0.1rc6.dev9+g69c3bed.d20220117
  -   - naparij: 0.2.2
  -   - scikit-image: 0.4.12
  -   - svg: 0.1.5'

and here is `conda list`
# Name                    Version                   Build  Channel
alabaster                 0.7.12                     py_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
asciitree                 0.3.3                      py_2    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
babel                     2.9.1              pyh44b312d_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
blas                      1.0                         mkl  
blosc                     1.21.0               h0e60522_0    conda-forge
bokeh                     2.4.2            py39hcbf5309_0    conda-forge
bottleneck                1.3.2            py39h7cc1a96_1  
brotli                    1.0.9                h8ffe710_6    conda-forge
brotli-bin                1.0.9                h8ffe710_6    conda-forge
brotlipy                  0.7.0           py39hb82d6ee_1003    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-blosc2                  2.0.4                h09319c2_1    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
cachey                    0.2.1              pyh9f0ad1d_0    conda-forge
certifi                   2021.10.8        py39hcbf5309_1    conda-forge
cffi                      1.15.0           py39h0878f49_0    conda-forge
cfitsio                   4.0.0                hd67004f_0    conda-forge
charls                    2.2.0                h39d44d4_0    conda-forge
charset-normalizer        2.0.10             pyhd8ed1ab_0    conda-forge
click                     8.0.3            py39hcbf5309_1    conda-forge
cloudpickle               2.0.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
cryptography              36.0.1           py39h7bc7c5c_0    conda-forge
cudatoolkit               11.5.0               hfde6d1b_9    conda-forge
cupy                      10.0.0           py39hc474d4a_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
cytoolz                   0.11.2           py39hb82d6ee_1    conda-forge
dask                      2021.12.0          pyhd8ed1ab_0    conda-forge
dask-core                 2021.12.0          pyhd8ed1ab_0    conda-forge
debugpy                   1.5.1            py39h415ef7b_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
distributed               2021.12.0        py39hcbf5309_0    conda-forge
docstring_parser          0.13               pyhd8ed1ab_0    conda-forge
docutils                  0.15.2           py39hcbf5309_3    conda-forge
entrypoints               0.3             pyhd8ed1ab_1003    conda-forge
fasteners                 0.16               pyhd8ed1ab_0    conda-forge
fastrlock                 0.8              py39h415ef7b_1    conda-forge
fonttools                 4.28.5           py39hb82d6ee_0    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freetype-py               2.2.0              pyh9f0ad1d_0    conda-forge
fsspec                    2021.11.1          pyhd8ed1ab_0    conda-forge
giflib                    5.2.1                h8d14728_2    conda-forge
heapdict                  1.0.1                      py_0    conda-forge
hsluv                     5.0.2              pyh44b312d_0    conda-forge
[... 151 lines cut ...]
traitlets                 5.1.1              pyhd8ed1ab_0    conda-forge
typing-extensions         4.0.1                hd8ed1ab_0    conda-forge
typing_extensions         4.0.1              pyha770c72_0    conda-forge
tzdata                    2021e                he74cb21_0    conda-forge
ucrt                      10.0.20348.0         h57928b3_0    conda-forge
urllib3                   1.26.8             pyhd8ed1ab_1    conda-forge
useq-schema               0.1.0                    pypi_0    pypi
vc                        14.2                 hb210afc_6    conda-forge
vispy                     0.9.4            py39h5d4886f_0    conda-forge
vs2015_runtime            14.29.30037          h902a5da_6    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
win32-setctime            1.0.4                    pypi_0    pypi
win_inet_pton             1.1.0            py39hcbf5309_3    conda-forge
wrapt                     1.13.3           py39hb82d6ee_1    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
[... cut ...]
zeromq                    4.3.4                h0e60522_1    conda-forge
zfp                       0.5.5                h0e60522_8    conda-forge
zict                      2.0.0                      py_0    conda-forge
zipp                      3.7.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.11            h8ffe710_1013    conda-forge
zmq                       0.0.0                    pypi_0    pypi
zstd                      1.5.1                h6255e5f_0    conda-forge

tlambert03, 2022-01-23:
> This may be a more general napari issue, because the napari buttons (e.g. layers buttons, toggle n-display, etc) also do not show any icons.

this definitely seems like it would be the case, but it's not immediately clear to me from your environment why this might be happening... if you're willing to play with this environment, could you update napari to 0.4.13?

ptbrown1729, 2022-01-23:
I updated napari, but this did not affect the behavior
`napari.sys_info()`
```
napari: 0.4.13
Platform: Windows-10-10.0.19041-SP0
Python: 3.9.9 | packaged by conda-forge | (main, Dec 20 2021, 02:36:06) [MSC v.1929 64 bit (AMD64)]
Qt: 5.12.9
PyQt5: 5.12.3
NumPy: 1.21.2
SciPy: 1.7.3
Dask: 2022.01.0
VisPy: 0.9.4

OpenGL:
  - GL version:  4.3.0 - Build 20.19.15.4531
  - MAX_TEXTURE_SIZE: 16384

Screens:
  - screen 1: resolution 1920x1080, scale 1.0

Plugins:
  - console: 0.0.4
  - micromanager: 0.0.1rc6.dev9+g69c3bed.d20220117
  - naparij: 0.2.2
  - scikit-image: 0.4.13
  - svg: 0.1.5
  ```

tlambert03, 2022-01-23:
OK thanks for checking. I'll have to try to find some time to reproduce.

ptbrown1729, 2022-01-23:
I also verified this issue does not have anything to do with this plugin ... I get the same behavior with the `naprij` plugin. So I will open this issue on napari instead

tlambert03, 2022-01-23:
thanks.  I'll meet you over there :)

tlambert03, 2022-01-23:
one thing I'll probably ask you to do over there as well is to create a new environment.  So, if you have a moment, try:

```
conda create -n test -c conda-forge python=3.9.9 napari
conda activate test
pip install napari-micromanager
napari -w micromanager
```

ptbrown1729, 2022-01-23:
https://github.com/napari/napari/issues/4004

tlambert03, 2022-01-23:
as mentioned over in the other thread.  It looks like this is specifically related to the qt distribution on conda-forge, and likely only on windows...  As a temporary workaround for anyone running into this, please create an environment where pyqt is _not_ sourced from conda forge:

```
conda create -n myenv python=3.9
conda activate myenv
pip install napari[all] napari-micromanager
napari -w micromanager
```

tlambert03, 2022-12-03:
now that we're not using `.ui` files anymore, I'm going to assume this is resolved (since it seemed to be specific to ui files, on a specific qt distribution, on windows).  If anyone sees this again please comment here

closed 2022-12-03 by tlambert03