pymmcore-plus/napari-micromanager #100: wrong checking option in mda

ianhi, 2022-02-17:
If you toggle the stage positoins check box in the multid widget you will get this error:

```
TypeError                                 Traceback (most recent call last)
File ~/mambaforge/envs/napmicr-test/lib/python3.9/site-packages/micromanager_gui/multid_widget.py:241, in MultiDWidget.toggle_checkbox_save_pos(self=)
    238     self.checkBox_save_pos.setEnabled(True)
    240 else:
--> 241     self.checkBox_save_pos.setCheckState(False)
        self.checkBox_save_pos = 
        self = 
    242     self.checkBox_save_pos.setEnabled(False)

TypeError: 'PySide2.QtWidgets.QCheckBox.setCheckState' called with wrong argument types:
  PySide2.QtWidgets.QCheckBox.setCheckState(bool)
Supported signatures:
  PySide2.QtWidgets.QCheckBox.setCheckState(PySide2.QtCore.Qt.CheckState)
```

which is these lines:

https://github.com/tlambert03/napari-micromanager/blob/db0e98bc53e293e826fc3e82473c9e210fbbccaa/micromanager_gui/multid_widget.py#L240-L242

attn: @fdrgsp

fdrgsp, 2022-02-18:
@ianhi Thanks! This might be the solution: `self.checkBox_save_pos.setCheckState(Qt.CheckState.Unchecked)`.
I've added that to PR #94 .

tlambert03, 2022-02-18:
Being a bug fix, this would be a nice thing to have in its own PR, so it doesn't need to get considered at the same time as a other new feature

fdrgsp, 2022-02-18:
> Being a bug fix, this would be a nice thing to have in its own PR, so it doesn't need to get considered at the same time as a other new feature

fix in #101

PR #101 (2022-02-18): fix setCheckState in mda using Qt.CheckState
Fix #100 

from bool: `self.checkBox_save_pos.setCheckState(False)`
to: `self.checkBox_save_pos.setCheckState(Qt.CheckState.Unchecked)`
closed 2022-02-18 by tlambert03