pymmcore-plus/napari-micromanager #106: New gui with superqt QCollapsibles

fdrgsp, 2022-02-25:
@tlambert03 @ianhi 

In relation to @tlambert03 suggestion in #90, I played a bit with the superqt `QCollapsible`.

In both I removed the camera `bit depth` and `binning` controls since I figured that we can set it through the `property browser` and we will be able to control it through the `group and preset table` (once ready). 

I also added the shutter widget in the gui (since I want to open a PR to implement shutters control)

What do you think? Which one of these you like better? (in the first one I used the collapsible only for the stages).

https://user-images.githubusercontent.com/70725613/155732262-33e706e5-5322-40fe-8bc0-2e318c4a33cc.mov

https://user-images.githubusercontent.com/70725613/155732279-df4e3f47-c43e-438b-9948-6b87e4cf0996.mov

tlambert03, 2022-02-25:
I think the space taken to make the 1-line Illumination and Camera widgets collapsible isn't worth it.  Perhaps is Objectives, shutters, and camera were all combined into a single widget, then _that_ 3-line widget should be collapsible.  but if it's a single row, it doesn't need collapse

fdrgsp, 2022-02-25:
> I think the space taken to make the 1-line Illumination and Camera widgets collapsible isn't worth it. Perhaps if Objectives, shutters, and camera were all combined into a single widget, then that 3-line widget should be collapsible. but if it's a single row, it doesn't need collapse

@tlambert03 You are right, It definitely makes more sense. What about now? 

https://user-images.githubusercontent.com/70725613/155738217-75208322-5827-48d1-b2db-15ad807b7ceb.mov

ianhi, 2022-02-25:
This looks awesome! I agree about the camera - and like the newest version

closed 2022-02-25 by fdrgsp