pymmcore-plus/napari-micromanager #112: GUI is disabled when opened programmatically

ianhi, 2022-02-28:
Due to the `_set_enabled` behavior if you modify the `launch-dev` script to have the config be loaded before creating the widget (a reasonable thing to do one when scripting)

```python
from pymmcore_plus import CMMCorePlus
from pathlib import Path

import napari
from useq import MDASequence

core = CMMCorePlus.instance()
core.loadSystemConfiguration(str(Path(__file__).parent / "tests" / "test_config.cfg"))

v = napari.Viewer()
dw, main_window = v.window.add_plugin_dock_widget("napari-micromanager")

sequence = MDASequence(
    channels=["Cy5", {"config": "FITC", "exposure": 50}],
    time_plan={"interval": 2, "loops": 5},
    z_plan={"range": 4, "step": 0.5},
    axis_order="tpcz",
    stage_positions=[(222, 1, 1), (111, 0, 0)],
)

main_window.mda.set_state(sequence)

napari.run()

```

then the entire GUI is still disabled.

I think the solution is to remove the top level concept of enabled/disabled and push that down to a per widget basis. They should check for the relevant loaded devices at creation and then listen the appropriate signals to (de)activate themselves as approriate.

tlambert03, 2022-02-28:
thanks.  agreed.  let's take another close look at `_set_enabled` and try to do away with it entirely

ianhi, 2022-02-28:
My proposed fix will also involve new signals being added. Which also opens the (maybe thorny) question of where do those signals go. see https://github.com/micro-manager/mmCoreAndDevices/issues/149

ianhi, 2022-02-28:
Relevant signals being added to core here: https://github.com/micro-manager/mmCoreAndDevices/pull/163

fdrgsp, 2023-07-28:
Closing since this is not happening anymore. Still have an unrelated issue already mentioned in #255.

```py
from pymmcore_plus import CMMCorePlus
from pathlib import Path

import napari
from useq import MDASequence

core = CMMCorePlus.instance()
core.loadSystemConfiguration(str(Path(__file__).parent / "tests" / "test_config.cfg"))

v = napari.Viewer()
dw, main_window = v.window.add_plugin_dock_widget("napari-micromanager")

sequence = MDASequence(
    channels=["Cy5", {"config": "FITC", "exposure": 50}],
    time_plan={"interval": 2, "loops": 5},
    z_plan={"range": 4, "step": 0.5},
    axis_order="tpcz",
    stage_positions=[(222, 1, 1), (111, 0, 0)],
)

main_window._show_dock_widget("MDA")
mda = v.window._dock_widgets.get("MDA").widget()
mda.set_state(sequence)

napari.run()
```

closed 2023-07-28 by fdrgsp