pymmcore-plus/napari-micromanager #127: Fixes for live mode

ianhi, 2022-03-03:
We can hold off on this for a bit till happy with #111 but I suspect we have some bugs in live mode.

See extensive explanation here: https://github.com/micro-manager/mmCoreAndDevices/issues/171#issuecomment-1058441397

In particular we need to:
1. Be even more careful than we currently are when updating the exposure
2. Not bother with passing a number to startContinuousSequenceAcquistion because it doesn't actually affect anything.

AlexCoul, 2022-06-29:
It's true, I was about to post an issue about that but I noticed this post, and I actually found a quick fix:
When I tried to use the Live mode, I had the error
```
TypeError                                 Traceback (most recent call last)
File ...\napari-micromanager\micromanager_gui\main_window.py:217, in MainWindow._start_live(self=<micromanager_gui.main_window.MainWindow object>)
    215 self.streaming_timer = QTimer()
    216 self.streaming_timer.timeout.connect(self.update_viewer)
--> 217 self.streaming_timer.start(self._mmc.getExposure())
        self.streaming_timer = None
        self = <micromanager_gui.main_window.MainWindow object at 0x000001372664B910>
        self._mmc = <CMMCorePlus at 0x13726666de0>

TypeError: arguments did not match any overloaded call:
  start(self, int): argument 1 has unexpected type 'float'
  start(self): too many arguments
```
so in `main_window.py` line 217 I changed `self.streaming_timer.start(self._mmc.getExposure())` to `self.streaming_timer.start(int(self._mmc.getExposure())))`, and now the Live mode works :-)

ianhi, 2022-06-30:
@AlexCoul would you like to open a PR with your fix?

AlexCoul, 2022-06-30:
Hi @ianhi, yep I just did!
As I said it's super small, but on my machine I can now use the live mode.

fdrgsp, 2023-07-28:
fixed in #181

closed 2023-07-28 by fdrgsp