pymmcore-plus/napari-micromanager #130: Update for new pymmcore-plus mda interface

tlambert03, 2022-03-05:
@ianhi, we need to update this repo now for https://github.com/tlambert03/pymmcore-plus/issues/101

ianhi, 2022-03-05:
Indeed. I actually got started on this but then it began morphing into a full overhaul of the mda widget. Maybe worth it to have a smaller part just updating the signals?

tlambert03, 2022-03-05:
Yeah, a bare minimum compatibility pr would be good.  How does the widget need to change?

ianhi, 2022-03-05:
I was hoping to make it more standalone like all the other PRs. It's also currently really coupled to the explore-sample widget (which actually handles the display) and some of it's logic lives on the main window rather than the widget

tlambert03, 2022-03-05:
ah, yes.  That would be excellent

tlambert03, 2023-07-28:
i'm not sure if there's anything actionable here anymore?  @ianhi?

ianhi, 2023-07-28:
I believe all is well here

closed 2023-07-28 by ianhi