pymmcore-plus/napari-micromanager #133: Move SequenceMETA in useq?

ianhi, 2022-03-05:
As I'm trying to more loosely couple the mda widget with the rest of the app I'm running into an issue with the sequenceMeta

https://github.com/tlambert03/napari-micromanager/blob/50ef8f4abe8885331fc49a468d00ecb4cec86f0d/micromanager_gui/multid_widget.py#L21-L28

maybe this is something that should be part of useq-schema? Or we should just use the existing metadata field there?

ianhi, 2022-03-05:
As a start see: https://github.com/tlambert03/useq-schema/pull/35

tlambert03, 2022-03-07:
Yes, this was definitely something we imagined putting upstream.  We just put it here while we figured out what we might need.  I'll comment over on your PR

ianhi, 2022-12-05:
Closed by https://github.com/pymmcore-plus/napari-micromanager/pull/224

closed 2022-12-05 by ianhi