pymmcore-plus/napari-micromanager #136: pixelSizeConfiguration considerations

tlambert03, 2022-03-06:
Opening this issue to get some feedback from @fdrgsp on the design goals we should be having for maintenance of pixel size configuration in general.  We should also compare/contrast with how this would be done in micromanager (there's always merit in maintaining consistency between the two, unless we decide that we can definitely offer something better).

From what I can tell, this is the goal

1) we offer an input somewhere that lets the user enter their camera pixel size.
2) when either the camera pixel size _or_ the objective changes, we:
   **a.** try to glean what the magnification of the objective is from the objective label (this is definitely something not done in micromanager, and probably error prone... might be better to have an explicit API rather than do the magic).
   **b.** define a new pixelSizeConfig entry for the objective
   **c.** `setPixelSizeUm` using the mag and camera pixel
    **d.** set the config as current...

I definitely like the general goal here, but I think we need an independent object, probably one that doesn't really need widgets, which monitors core property changes for the objective and whatever means we use to retrieve camera pixel size.

I also don't think we need that pixel size entry to be part of the main window.

fdrgsp, 2022-03-06:
I definitely agree with that! I created the `px_size spinbox` at the beginning only to test the sample explorer widget. So I think we can remove that and for now rely only on setting the px size info in the system configuration file.

> retrieve camera pixel size.

how would you suggest to do that if there is no initial px size info provided by the system configuration file? (I'd love to get the camera px size info from the micromanager camera properties but there I don't think there is this info).

tlambert03, 2022-03-07:
> how would you suggest to do that if there is no initial px size info provided by the system configuration file? 

not sure.  I'm not opposed to having an input widget dialog somewhere.  But don't think it needs a permanent home in main window.  I'm adding a `Micro-Manager` menu in the #131, so perhaps we could add it as a command up there.

I'm going to merge #119 now, so explorer may break temporarily... but I definitely want to get it working again ASAP cause it's very cool.  Let's sync up and discuss how best to collect the necessary info

fdrgsp, 2023-07-28:
Done in `pymmcore-widgets` `PixelSizeWidget`.

closed 2023-07-28 by fdrgsp