pymmcore-plus/napari-micromanager #151: core_widget vs gui_object

ianhi, 2022-03-14:
What's the line between these two? In #150 I made the exposure widget a core_widget because it's following a core property. So maybe core_widget handle the interactions with core, and gui_objects are responsible for creating and laying out those widgets?

tlambert03, 2022-03-14:
`core_widgets` is basically what I imagine breaking out into a new package soon.  So, if it simply provides a Qt view onto the core model and could generally be used by someone else composing a GUI for mmcore, then it can go in code_widget (and will eventually get plucked out).
everything that is our opinionated composition of these widgets into "the napari micromanager interface" should go outside of that folder

ianhi, 2022-04-28:
> basically what I imagine breaking out into a new package soon

It might also be helpful to have some of the precomposed sets (e.g. XYZ stage widgets) available as part of that package as well as those would probably also be commonly useful.

Also, thoughts on a threshold for doing something like this? After talking to @easunarunachalam today it seems that these widgets would be really helpful for him. As I understand it live mode image display is locked into a proprietary program, so it's helpful to have the controls objects + gui widgets for custom motors/laser controls as a floating window without any displays.

tlambert03, 2022-12-03:
closing this.  We now have https://github.com/pymmcore-plus/pymmcore-widgets

we still have an internal `_gui_objects` module, which is mostly for stuff that needs slight tweaking for the purpose of napari-micromanager.  But people should generally look to pymmcore-widgets if they'd like to compose some of the gui (but not necessarily use it exactly as napari-micromanager has done)

closed 2022-12-03 by tlambert03