pymmcore-plus/napari-micromanager #178: Napari + Napari Micromanager plugin too large for my screen

CedricEsp, 2022-06-22:
Hello,

I have a small issue on a MacBook Pro (13inches), the napari-micromanager main window is too large for my screen, and unless I float the panel, I can't make the napari window smaller. Basically I don't have access to the bottom of the page:
[image]
Not sure if there is something I could do on my side, outside accepting it :) 

Thanks!

tlambert03, 2022-06-22:
Thanks @CedricEsp, I agree we generally need to make the main window more compact and adjustable.  Perhaps a workaround in the meantime: you can collapse the “stages” and “microscope” sections by clicking on their titles.  That should recover some space

ianhi, 2022-06-22:
I suspect this is more about the horizontal space than the vertical. You can pop the plugin out into a standalone window using one of these buttons: 
[image]
 which gives the horizontal space back to the image.

fdrgsp, 2023-07-28:
closing. The new GUI has individual widget in the toolbar.

closed 2023-07-28 by fdrgsp