pymmcore-plus/napari-micromanager #186: Linux Support

RileyMShea, 2022-07-09:
I have run the full pytest suite under the latest `branch` in Ubuntu Linux and everything passes.

## Notes

The main wrinkle with linux is that micro-manager or at least mmcore need to be compiled from source, but it was relatively straightforward with `conda` and `apt` on Ubuntu 22.04. https://github.com/micro-manager/mmCoreAndDevices#building-on-mac-and--linux

The only other thing I had to get all the tests to pass was to copy the `MMConfig_demo.cfg` to `/usr/local/lib/micro-manager`.

I believe it should be possible either via [a github actions caching directive](https://stackoverflow.com/a/59277514/7262049) to cache the compilation of micro-manager/mmcore, so it re-compilation happens on new release of mmcore or some other infrequent heuristic.

---
I'm willing to put a PR together that integrates an Ubuntu runner into the github actions matrix.

tlambert03, 2022-07-10:
> I'm willing to put a PR together that integrates an Ubuntu runner into the github actions matrix.

that would be fantastic, and very much appreciated!  Thanks @RileyMShea.

ianhi, 2022-07-10:
> The main wrinkle with linux is that micro-manager or at least mmcore need to be compiled from source, but it was relatively straightforward with conda and apt on Ubuntu 22.04.

It turns out the only thing you actually need from this compilation is the compiled device adapters - we get `mmcore` via `pymmcore`. However, the build system is currently such that there is no way to build them without also building all of micromanager (this will hopefully change in the future, see https://github.com/micro-manager/micro-manager/issues/1392#issuecomment-1033140970). Furthermore we don't even need all the device adapter since we only use the demo camera in the tests. All in I think we only need to cache the files for the demo camera device driver which on my system are:
```
libmmgr_dal_DemoCamera.la
libmmgr_dal_DemoCamera.so.0
```

 
> I'm willing to put a PR together that integrates an Ubuntu runner into the github actions matrix.

:tada: :tada: yes please :)  We can then use the infrastructure from that to set up the same for https://github.com/tlambert03/pymmcore-plus

--- 

@RileyMShea are you planning on using this on linux to drive a real microscope? If so there are some important caveats about what hardware may work see: https://github.com/micro-manager/mmCoreAndDevices/pull/86#issuecomment-984033653
tldr: currently non of the closed source adapters, long term don't expect all the closed source versions to build.

RileyMShea, 2022-07-11:
@ianhi Hm so sounds like I'll need to build everything but only need to cache those two files.  Will try to prototype it in docker then adapt it to gh actions once the docker prototype is operational.

> @RileyMShea are you planning on using this on linux to drive a real microscope?

Not atm. The microscope's I will eventually have to work with will be running on a windows machine, but I figured for development it would be easiest to just work on my main Ubuntu machine since that's where I feel strongest.

I will Keep in mind your note about limited hardware support though should I decide to try a real camera on linux though.

ianhi, 2022-07-11:
> but I figured for development it would be easiest to just work on my main Ubuntu machine since that's where I feel strongest.

Makes sense. FWIW I do all my development on my PopOS laptop and after getting the device drivers built haven't run in to any issues.

Out of curiosity what are you planning on using this for? If there's any sort of bespoke acquisitions you want to do I can point you to some examples and issues with discussion on how that haven't made it into proper documentation yet.

tlambert03, 2022-12-03:
closing thanks to @RileyMShea's PR #191

closed 2022-12-03 by tlambert03