pymmcore-plus/napari-micromanager #189: Stop "live acquisition"

CedricEsp, 2022-07-11:
Hello,

Quick question, I wonder if there is a "clean" to stop the live acquisition programmatically? 
Doing `core.stopSequenceAcquisition(core.getCameraDevice())` does stop the sequence but doesn't toggle the `Live` button to `Stop` and I can't clear the preview layer in napari.

ianhi, 2022-07-11:
Huh that's surprising, I would have suspected that this line would have made it all work: https://github.com/pymmcore-plus/napari-micromanager/blob/main/micromanager_gui/main_window.py#L90

Should be investigated, but as a workaround just call `main_window._stop_live()`

fdrgsp, 2023-07-28:
I tested it again and this is solved in `pymmcore-widgets` `LiveButton`.

closed 2023-07-28 by fdrgsp