pymmcore-plus/napari-micromanager #192: Rotate / Flip live image

CedricEsp, 2022-07-18:
Hello,

The camera chip I am using is not square and I would like to Flip and rotate the image, I can do that after acquisition in the engine but I wonder if there is an option to rotate the image on the live view as well? 

Thanks!
Cedric

ianhi, 2022-07-19:
Curious what @tlambert03 thinks but I think I'd prefer to implement such functionality in `pymmcore-plus` instead of here. Something like:

```python
core.image_transforms.append(np.flip)
```

tlambert03, 2022-07-19:
> Curious what @tlambert03 thinks but I think I'd prefer to implement such functionality in pymmcore-plus instead of here. Something like:

love that idea.  Maybe it's time to generally attack adding functional hooks to the engine?

ianhi, 2022-08-08:
> love that idea. Maybe it's time to generally attack adding functional hooks to the engine?

When you say "functional hooks" what does that mean?

tlambert03, 2022-08-08:
Meaning: it's one thing to be able to connect callbacks to events emitted by the MDA... but that's one way communication: we can't modify the events and or the image  in the process.  A functional hook would provide a way to modify either the event object and or the image itself before further processing and/or event emission

ianhi, 2022-08-08:
Ahh gotcha. Is this something that we should use a library (a la psygnal) for, or is better to go simpler with something like keeping a list of functions to evaluate?