pymmcore-plus/napari-micromanager #199: Shutters and channel setting

MasafumiTsuboi, 2022-09-27:
Hi,

I have 2 problems with shutter and channel setting.
・In the main window, the shutter button in microscope is grayed out and can't be selected.
・I can't add channels in the Snap/Live or multi-D acquisition tabs.

Micro-manager configuration worked in imageJ, but it didn't work as a plug-in tool on the Napari platform.

Is there any reason why this would be the case?

Thanks for all your help!

tlambert03, 2022-09-27:
Did you load your micromanager configuration in the widget at the top right?

MasafumiTsuboi, 2022-09-27:
Thanks for your reply! I loaded my micromanager configuration, and some functions (laser, lens, etc.) did work. Is this the problem on the configuration setting?

fdrgsp, 2022-09-27:
@MasafumiTsuboi Hi! Can you please post here the content of your micromanager configuration file?

In general, the `channel widget` works only if you have some channel cfg already defined in the micromanager configuration file (maybe called "Channels").

About the shutters, in the current release there is no shutter control (that's why the widget is grayed out). We will add more functionalities/widgets in the next release. If you want to test the shutter widgets you can install `napari-micromanager` using the line below (I suggest you to create a new environment):

`pip install napari-micromanager@git+https://github.com/fdrgsp/napari-micromanager.git@with_shutter`

MasafumiTsuboi, 2022-09-28:
Thank you for your reply! Here is my configuration file. I will install the micromanager with shutter version. Thank you again for your kind help!
[MMConfig_orca_CSUW1.cfg.zip](https://github.com/pymmcore-plus/napari-micromanager/files/9660532/MMConfig_orca_CSUW1.cfg.zip)

ianhi, 2023-01-19:
https://github.com/pymmcore-plus/pymmcore-widgets/pull/121

should also help here - shutters not modified by autofocus will be able to be manually controlled

fdrgsp, 2023-07-28:
closing. fixed in https://github.com/pymmcore-plus/pymmcore-widgets/pull/121.

closed 2023-07-28 by fdrgsp