pymmcore-plus/napari-micromanager #203: Keeping LED on during z stack

CedricEsp, 2022-10-11:
Hi,
I don't know if it's a micro-manager question or a napari-micro/pymmcore question but when I acquire a z-stack, every step, the LED is turn on and off which slightly slow down the acquisition. Is there a way to keep the LED on?  Or is it something I have to setup in micro-manager?

Thanks!

tlambert03, 2022-10-12:
I believe, (but am not sure/have not tested) that you would need to use [`mmc.setAutoShutter(False)`](https://valelab4.ucsf.edu/~MM/doc/MMCore/html/class_c_m_m_core.html#a4044c7fe6f58db4ca69c0bb4016f3635), and then manually open/close the shutter before/after the Z-stack.

https://github.com/pymmcore-plus/useq-schema/issues/38 would simply be the declarative form of that (i.e. _instructing_ an mda engine to do so, but not actually telling it how to).

i think short term solution would require you to modify the pymmcore-plus engine  (agree @ianhi?)

"ideal" long-term steps here would be:
1. add the directive to useq-schema somewhere
2. honor the directive within our default mda engine

CedricEsp, 2022-10-12:
Thanks @tlambert03 it works very well, however it doesn't seem to be the reason why the z-stack was quite "slow".
So if I run something like that:
```python
for i, z_pos in enumerate(z_stack_pos):
            
            self._mmc.setPosition(zdev, float(z_pos))
            #need to wait for the device or it doesn't work properly
            self._mmc.waitForDevice(zdev)
            self._mmc.snapImage()
            data[i, :, :] = self._mmc.getImage()
```
It goes extremely fast
While if do the same z stack using the `multi-D acquisition`, it goes significantly slower (sorry I don't have exact number), is there a particular reason for that? 

Thanks!

ianhi, 2022-10-12:
> i think short term solution would require you to modify the pymmcore-plus engine (agree @ianhi?)

I think so.

> While if do the same z stack using the multi-D acquisition, it goes significantly slower (sorry I don't have exact number), is there a particular reason for that?

I wonder if some of the callbacks hooked up to the `emit` are slowing you down. Can you try running that loop with napari closed? That way you could see if the slowness is coming from napari-micromanager or something in the engine itself.

tlambert03, 2022-10-12:
Another thing to mention here is that we haven't optimized for "sequenceable" hardware in pymmcore plus's mda engine yet.  

I'm curious: Are you by chance comparing to the speed you see in micromanager itself?  I would, in some cases, still expect it to be faster than pymmcore-plus

tlambert03, 2022-10-12:
Great!  I'll take that 😂

tlambert03, 2022-12-03:
closing this since it seems to be mostly resolved.  the ultimate underlying issue here is https://github.com/pymmcore-plus/useq-schema/issues/38

closed 2022-12-03 by tlambert03