pymmcore-plus/napari-micromanager #204: core.waitForSystem() doesn't seem to be waiting

CedricEsp, 2022-11-01:
Hello,

I think I mentioned that before but it seems that the `core.waitForSystem() ` doesn't seem to be waiting enough. For example, If I start from the load position on a stage and start an experiment, it would start the acquisition before the stage is all the way down. However, if I add a ` time.sleep(2.2)`  after `self._prep_hardware(event)` it works fine.

ianhi, 2022-11-01:
what stage do you have? That sounds like an issue in the driver code

CedricEsp, 2022-11-01:
I am running the focus drive on an AxioImager Z2. Is there a way to check if it's the driver code?

ianhi, 2022-11-01:
I'm pretty sure the relevant code would live here: https://github.com/micro-manager/mmCoreAndDevices/tree/main/DeviceAdapters/ZeissAxioZoom

As a test can you simplify the system to just do the following:

```python
core.set_position(0) # assuming this is a safe position for your system
core.set_position(3000)
core.waitForSystem()
print(core.get_position())
```

CedricEsp, 2022-11-04:
Indeed the` core.waitForSystem()` is definitely not waiting for the system - Is there something I could try or should are raise the issue to micromanager directly?

ianhi, 2022-11-05:
You should raise it here: https://github.com/micro-manager/mmCoreAndDevices/issues

fdrgsp, 2023-07-28:
Closing here. Issue opened in https://github.com/micro-manager/mmCoreAndDevices/issues/297

closed 2023-07-28 by fdrgsp