pymmcore-plus/napari-micromanager #261: Bug: More issues with `event_filter`

ianhi, 2023-05-09:
#254 doesn't seem to have fully fixed things for me yet unfortunately.

When I launch napari-micromanager from a script:

```python
import napari
from pymmcore_plus import CMMCorePlus

v = napari.Viewer()
dw, main_window = v.window.add_plugin_dock_widget("napari-micromanager")

core = CMMCorePlus.instance()
core.loadSystemConfiguration()

napari.run()
```

then it freezes multiple times as everything is loaded.

ianhi, 2023-05-09:
accidentally hit enter before I meant to post so many edits above - but @tlambert03 do you also see this and/or have any insight into potentially lessening the impact of the event filter

tlambert03, 2023-05-09:
we should ask @fdrgsp ... he's been using it much more than I have.

ianhi, 2023-05-10:
per a tip from @fdrgsp on zulip using `pyside2` seems to fix it. Though ideally it should work with PyQt5 also :/

tlambert03, 2023-05-10:
ok will have a look

tlambert03, 2023-05-12:
I've been unable to reproduce this on macOS 13 with
```
napari: 0.4.17
Platform: macOS-13.3.1-x86_64-i386-64bit
System: MacOS 13.3.1
Python: 3.10.11 | packaged by conda-forge | (main, May 10 2023, 19:07:22) [Clang 14.0.6 ]
Qt: 5.15.2
PyQt5: 5.15.9
```

the event filter is only there to force the toolbar to be on the top.  So, if you want to add an environment variable to disable it, or anything else you want to do to make it work better for you, it's fine with me.

The event filter being called a gazillion times isn't in itself too surprising (any and all events pass like wildfire through the widget tree...).  But it should almost never hit that conditional: `event.type() == QEvent.Type.Move and obj is self`...  (though of course simply evaluating that could perhaps be the problem?  not sure why it would be os specific)

tlambert03, 2023-07-28:
@ianhi, please recheck.  Since I was unable to reproduce even on PyQt5, I'm inclined to close this, but still waiting to hear from you

ianhi, 2023-07-28:
I can still reproduce this on python 3.11. It resolves if I `export QT_API=pyside2` before running.

ianhi, 2023-07-28:
Since I can reproduce I'm also happy to spend a bit of time on this next time I come by HMS

tlambert03, 2023-07-28:
that would be great too :)

ianhi, 2023-08-29:
solved by #280

closed 2023-08-29 by ianhi