pymmcore-plus/napari-micromanager #279: napari-micromanager crashing napari on python 3.11

ptbrown1729, 2023-08-25:
I created a fresh python 3.11 environment to run napari-micromanager, but when trying to run napari micromanager it crashes with a hard to interpret error (looks like something on the QT side?). Does anyone have any insight/ideas how to debug what is going on?

I create the environment on a Windows 10 enterprise machine using mamba
```
conda create -n wf_ctrl2 python=3.11
conda activate wf_ctrl2
git clone https://github.com/tlambert03/napari-micromanager
cd napari-micromanager
pip install -e .
conda install -c conda-forge pyqt
python
```

Then with python I run
```
import napari
v = napari.Viewer()
dw, main_window = v.window.add_plugin_dock_widget("napari-micromanager")
```

This opens the Napari window briefly, but then it crashes. Here is the output:
```
0.00s - Debugger warning: It seems that frozen modules are being used, which may
0.00s - make the debugger miss breakpoints. Please pass -Xfrozen_modules=off
0.00s - to python to disable frozen modules.
0.00s - Note: Debugging will proceed. Set PYDEVD_DISABLE_FILE_VALIDATION=1 to disable this validation.
WARNING: QMainWindowLayout::count: ?
```

output of `pip list`
```
alabaster                     0.7.13
app-model                     0.2.0
appdirs                       1.4.4
asciitree                     0.3.3
asttokens                     2.2.1
attrs                         23.1.0
Babel                         2.12.1
backcall                      0.2.0
build                         0.10.0
cachey                        0.2.1
certifi                       2023.7.22
charset-normalizer            3.2.0
click                         8.1.7
cloudpickle                   2.2.1
colorama                      0.4.6
comm                          0.1.4
dask                          2023.8.1
debugpy                       1.6.7.post1
decorator                     5.1.1
docstring-parser              0.15
docutils                      0.17.1
entrypoints                   0.4
executing                     1.2.0
fasteners                     0.18
fonticon-materialdesignicons6 6.9.96
freetype-py                   2.4.0
fsspec                        2023.6.0
HeapDict                      1.0.1
hsluv                         5.0.3
idna                          3.4
imageio                       2.31.1
imagesize                     1.4.1
importlib-metadata            6.8.0
in-n-out                      0.1.8
ipykernel                     6.25.1
ipython                       8.14.0
ipython-genutils              0.2.0
jedi                          0.19.0
Jinja2                        3.1.2
jsonschema                    4.19.0
jsonschema-specifications     2023.7.1
jupyter_client                8.3.0
jupyter_core                  5.3.1
kiwisolver                    1.4.5
lazy_loader                   0.3
locket                        1.0.0
loguru                        0.7.0
magicgui                      0.7.3
markdown-it-py                3.0.0
MarkupSafe                    2.1.3
matplotlib-inline             0.1.6
mdurl                         0.1.2
mypy-extensions               1.0.0
napari                        0.4.18
napari-console                0.0.8
napari-micromanager           0.0.1rc8.dev61+g433cb9e C:\Users\qi2lab\Documents\napari-micromanager
napari-plugin-engine          0.2.0
napari-svg                    0.1.10
nest-asyncio                  1.5.7
networkx                      3.1
npe2                          0.7.2
numcodecs                     0.11.0
numpy                         1.25.2
numpydoc                      1.5.0
packaging                     23.1
pandas                        2.0.3
parso                         0.8.3
partd                         1.4.0
pickleshare                   0.7.5
Pillow                        10.0.0
Pint                          0.22
pip                           23.2.1
platformdirs                  3.10.0
ply                           3.11
pooch                         1.7.0
prompt-toolkit                3.0.39
psutil                        5.9.5
psygnal                       0.9.3
pure-eval                     0.2.2
pydantic                      1.10.12
pydantic-compat               0.0.4
Pygments                      2.16.1
pymmcore                      10.4.0.71.0
pymmcore-plus                 0.8.0
pymmcore-widgets              0.4.1
PyOpenGL                      3.1.7
pyproject_hooks               1.0.0
PyQt5                         5.15.9
PyQt5-sip                     12.12.2
python-dateutil               2.8.2
pytz                          2023.3
PyWavelets                    1.4.1
pywin32                       306
PyYAML                        6.0.1
pyzmq                         25.1.1
qtconsole                     5.4.3
QtPy                          2.3.1
referencing                   0.30.2
requests                      2.31.0
rich                          13.5.2
rpds-py                       0.9.2
scikit-image                  0.21.0
scipy                         1.11.2
setuptools                    68.1.2
sip                           6.7.11
six                           1.16.0
snowballstemmer               2.2.0
Sphinx                        4.5.0
sphinxcontrib-applehelp       1.0.4
sphinxcontrib-devhelp         1.0.2
sphinxcontrib-htmlhelp        2.0.1
sphinxcontrib-jsmath          1.0.1
sphinxcontrib-qthelp          1.0.3
sphinxcontrib-serializinghtml 1.1.5
stack-data                    0.6.2
superqt                       0.5.3
tifffile                      2023.8.12
toml                          0.10.2
tomli                         2.0.1
tomli_w                       1.0.0
toolz                         0.12.0
tornado                       6.3.3
tqdm                          4.66.1
traitlets                     5.9.0
typer                         0.9.0
typing_extensions             4.7.1
tzdata                        2023.3
urllib3                       2.0.4
useq-schema                   0.4.6
vispy                         0.12.2
wcwidth                       0.2.6
wheel                         0.41.2
win32-setctime                1.1.0
wrapt                         1.15.0
zarr                          2.16.1
zipp                          3.16.2
```

ptbrown1729, 2023-08-25:
It seems this is some problem with the current main branch, as I don't have this problem if I install with the PyPI version with
`pip install napari-micromanager`

I have the same problem with the main branch for several other versions
* python 3.10 
* python 3.10 without editable install option
* python 3.10 with napari 0.4.17

with python 3.10 and napari 0.4.16 or 0.4.15 I get a different error:
```
WARNING: QMainWindowLayout::count: ?
>>> WARNING: QWindowsWindow::setGeometry: Unable to set geometry 1728x771+520+265 (frame: 1744x810+512+234) on QWidgetWindow/"_QtMainWindowClassWindow" on "\\.\DISPLAY1". Resulting geometry: 1684x771+520+265 (frame: 1700x810+512+234) margins: 8, 31, 8, 8 minimum size: 1119x771 MINMAXINFO maxSize=0,0 maxpos=0,0 mintrack=1135,810 maxtrack=0,0)
```

tlambert03, 2023-08-27:
Hey @ptbrown1729 👋

going to try to reproduce now... but one thing I immediately see is that you've cloned my personal fork rather than this repo:

```
git clone https://github.com/tlambert03/napari-micromanager
```

should be 

```
https://github.com/pymmcore-plus/napari-micromanager.git
```

looks like my `main` is a number of commits behind.

That said... I've been seeing the warnings you mentioned above for a while as well, but sort of assumed they were napari issues in general (and not specific to napari-micromanager).  I hadn't dug any deeper, however, because I haven't ever actually seen it hard crash back to the terminal as you mention in your first post.  Will let you know if I find anything

tlambert03, 2023-08-27:
ok, I was able to reproduce with both my personal fork and with main here...

The first warning about frozen modules happens simply on creating a napari window, and has also been mentioned in the napari repo: https://github.com/napari/napari/issues/5514#issuecomment-1408869123

the `WARNING: QMainWindowLayout::count: ?` happens when creating the actual plugin window, and shortly thereafter I also get a crash (though, for what it's worth, I'm only testing in a remote vscode environment without a display...)

I can also create a main window here just fine... it's only the adding of the dock widget to napari that seems to be the problem:

```python
import napari
from napari_micromanager import MainWindow

v = napari.Viewer()  # napari warning
mm = MainWindow(v)  # fine

v.window.add_dock_widget(win)  # WARNING: QMainWindowLayout::count: ? and crash
```

lemme see if I can figure out what's happening

tlambert03, 2023-08-27:
can also repro on macos

tlambert03, 2023-08-27:
also linking https://github.com/napari/napari/issues/5815 ... since @ianhi saw the same thing there and narrowed it down to the eventFilter we're installing, and observed that it only happened for pyqt5, but not pyside.

tlambert03, 2023-08-27:
the `WARNING: QMainWindowLayout::count: ?` warning comes simply because we use QMainWindow instead of QWidget... @fdrgsp, we should probably change that to `QWidget` (can't remember if we used the menu capabilities of QMainWindow... but we shouldn't use it for the napari plugin part)

tlambert03, 2023-08-27:
fix incoming...

ptbrown1729, 2023-08-28:
Oops, I was intending to clone main from the `pymmcore-plus/napari-micgromanager`. Looks like the link from PyPi took me to your fork. But it sounds like it was an issue in the main repo as well. Thanks for the quick response!

closed 2023-08-29 by tlambert03

tlambert03, 2023-08-29:
>  Looks like the link from PyPi took me to your fork

ah yep, looks like we haven't released to PyPI in quite a while.  It's updated in pyproject, but a new release is needed.  will do soonish