pymmcore-plus/napari-micromanager #286: AttributeError: 'method-wrapper' object has no attribute '__module__'

mshnschnko, 2023-09-13:
I have this error when i try to activate plugin 'napari-live-recording'. I asked such a question in the plugin repository, but there is no answer yet. Maybe someone has faced a similar problem.

```---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
File ~\spbstu_projects\napari_viewer\.venv\Lib\site-packages\napari\_qt\menus\plugins_menu.py:105, in PluginsMenu._add_plugin_actions.<locals>._add_toggle_widget(key=('napari-live-recording', 'Live recording'), hook_type='dock')
    102     return
    104 if hook_type == 'dock':
--> 105     self._win.add_plugin_dock_widget(*key)
        key = ('napari-live-recording', 'Live recording')
        self._win = <napari._qt.qt_main_window.Window object at 0x00000254AEE3A210>
        self = <napari._qt.menus.plugins_menu.PluginsMenu object at 0x00000254AEDFF9A0>
    106 else:
    107     self._win._add_plugin_function_widget(*key)

File ~\spbstu_projects\napari_viewer\.venv\Lib\site-packages\napari\_qt\qt_main_window.py:811, in Window.add_plugin_dock_widget(self=<napari._qt.qt_main_window.Window object>, plugin_name='napari-live-recording', widget_name='Live recording', tabify=False)
    808         wdg = wdg._magic_widget
    809     return dock_widget, wdg
--> 811 wdg = _instantiate_dock_widget(
        Widget = <class 'napari_live_recording.NapariLiveRecording'>
        self = <napari._qt.qt_main_window.Window object at 0x00000254AEE3A210>
    812     Widget, cast('Viewer', self._qt_viewer.viewer)
    813 )
    815 # Add dock widget
    816 dock_kwargs.pop('name', None)

File ~\spbstu_projects\napari_viewer\.venv\Lib\site-packages\napari\_qt\qt_main_window.py:1465, in _instantiate_dock_widget(wdg_cls=<class 'napari_live_recording.NapariLiveRecording'>, viewer=Viewer(camera=Camera(center=(0.0, 255.5, 255.5),...ouse_drag_gen={}, _mouse_wheel_gen={}, keymap={}))
   1460             break
   1461         # cannot look for param.kind == param.VAR_KEYWORD because
   1462         # QWidget allows **kwargs but errs on unknown keyword arguments
   1463 
   1464 # instantiate the widget
-> 1465 return wdg_cls(**kwargs)
        kwargs = {'napari_viewer': Viewer(camera=Camera(center=(0.0, 255.5, 255.5), zoom=1.3993706995412845, angles=(0.0, 0.0, 90.0), perspective=0.0, mouse_pan=True, mouse_zoom=True), cursor=Cursor(position=(485.24613920433933, 565.639420545863), scaled=True, size=1, style=<CursorStyle.STANDARD: 'standard'>), dims=Dims(ndim=2, ndisplay=2, last_used=0, range=((0.0, 512.0, 1.0), (0.0, 512.0, 1.0)), current_step=(255, 255), order=(0, 1), axis_labels=('0', '1')), grid=GridCanvas(stride=1, shape=(-1, -1), enabled=False), layers=[<Image layer 'preview' at 0x254af02d290>], help='use <2> for transform', status={'layer_base': 'preview', 'source_type': '', 'plugin': '', 'coordinates': ' [485 566]'}, tooltip=Tooltip(visible=False, text=''), theme='dark', title='napari', mouse_over_canvas=True, mouse_move_callbacks=[], mouse_drag_callbacks=[], mouse_double_click_callbacks=[], mouse_wheel_callbacks=[<function dims_scroll at 0x00000254AA47D120>], _persisted_mouse_event={}, _mouse_drag_gen={}, _mouse_wheel_gen={}, keymap={})}
        wdg_cls = <class 'napari_live_recording.NapariLiveRecording'>

File ~\spbstu_projects\napari_viewer\.venv\Lib\site-packages\napari_live_recording\__init__.py:11, in NapariLiveRecording.__init__(self=<napari_live_recording.NapariLiveRecording object>, napari_viewer=Viewer(camera=Camera(center=(0.0, 255.5, 255.5),...ouse_drag_gen={}, _mouse_wheel_gen={}, keymap={}))
      9 self.app = QApplication.instance()
     10 self.mainController = MainController()
---> 11 self.anchor = ViewerAnchor(napari_viewer, self.mainController)
        self.mainController = <napari_live_recording.control.MainController object at 0x00000254AEE5FB50>
        self = <napari_live_recording.NapariLiveRecording object at 0x00000254AEE4B7F0>
        napari_viewer = Viewer(camera=Camera(center=(0.0, 255.5, 255.5), zoom=1.3993706995412845, angles=(0.0, 0.0, 90.0), perspective=0.0, mouse_pan=True, mouse_zoom=True), cursor=Cursor(position=(485.24613920433933, 565.639420545863), scaled=True, size=1, style=<CursorStyle.STANDARD: 'standard'>), dims=Dims(ndim=2, ndisplay=2, last_used=0, range=((0.0, 512.0, 1.0), (0.0, 512.0, 1.0)), current_step=(255, 255), order=(0, 1), axis_labels=('0', '1')), grid=GridCanvas(stride=1, shape=(-1, -1), enabled=False), layers=[<Image layer 'preview' at 0x254af02d290>], help='use <2> for transform', status={'layer_base': 'preview', 'source_type': '', 'plugin': '', 'coordinates': ' [485 566]'}, tooltip=Tooltip(visible=False, text=''), theme='dark', title='napari', mouse_over_canvas=True, mouse_move_callbacks=[], mouse_drag_callbacks=[], mouse_double_click_callbacks=[], mouse_wheel_callbacks=[<function dims_scroll at 0x00000254AA47D120>], _persisted_mouse_event={}, _mouse_drag_gen={}, _mouse_wheel_gen={}, keymap={})
     12 self.setLayout(self.anchor.mainLayout)
     13 self.app.lastWindowClosed.connect(self.on_close_callback)

File ~\spbstu_projects\napari_viewer\.venv\Lib\site-packages\napari_live_recording\ui\__init__.py:40, in ViewerAnchor.__init__(self=<napari_live_recording.ui.ViewerAnchor object>, napari_viewer=Viewer(camera=Camera(center=(0.0, 255.5, 255.5),...ouse_drag_gen={}, _mouse_wheel_gen={}, keymap={}), mainController=<napari_live_recording.control.MainController object>)
     38 self.selectionWidget.setDeviceSelectionWidget(list(devicesDict.keys()))
     39 self.selectionWidget.setAvailableCameras(list(devicesDict.keys()))
---> 40 self.recordingWidget = RecordHandling()
        self = <napari_live_recording.ui.ViewerAnchor object at 0x00000254B9B609D0>
        RecordHandling = <class 'napari_live_recording.ui.widgets.RecordHandling'>
     41 verticalSpacer = QSpacerItem(0, 1, QSizePolicy.Minimum, QSizePolicy.Minimum)
     42 self.mainLayout.addWidget(self.selectionWidget.group)

File ~\spbstu_projects\napari_viewer\.venv\Lib\site-packages\napari_live_recording\ui\widgets.py:397, in RecordHandling.__init__(self=<napari_live_recording.ui.widgets.RecordHandling object>)
    394 self.layout = QGridLayout()
    396 self.formatLabel = QLabel("File format")
--> 397 self.formatComboBox = QEnumComboBox(enum_class=FileFormat)
        self = <napari_live_recording.ui.widgets.RecordHandling object at 0x00000254AEE7C9D0>
        QEnumComboBox = <class 'superqt.combobox._enum_combobox.QEnumComboBox'>
    398 self.formatLabel.setAlignment(Qt.AlignmentFlag.AlignCenter)
    400 self.folderTextEdit = QLineEdit(
    401     os.path.join(os.path.expanduser("~"), "Documents")
    402 )

File ~\spbstu_projects\napari_viewer\.venv\Lib\site-packages\superqt\combobox\_enum_combobox.py:42, in QEnumComboBox.__init__(self=<superqt.combobox._enum_combobox.QEnumComboBox object>, parent=None, enum_class=<enum 'FileFormat'>, allow_none=False)
     40 self._allow_none = False
     41 if enum_class is not None:
---> 42     self.setEnumClass(enum_class, allow_none)
        self = <superqt.combobox._enum_combobox.QEnumComboBox object at 0x00000254AEE7CC10>
        enum_class = <enum 'FileFormat'>
        allow_none = False
     43 self.currentIndexChanged.connect(self._emit_signal)

File ~\spbstu_projects\napari_viewer\.venv\Lib\site-packages\superqt\combobox\_enum_combobox.py:53, in QEnumComboBox.setEnumClass(self=<superqt.combobox._enum_combobox.QEnumComboBox object>, enum=<enum 'FileFormat'>, allow_none=False)
     51     super().addItem(NONE_STRING)
     52 names = map(_get_name, self._enum_class.__members__.values())
---> 53 _names = dict.fromkeys(names)  # remove duplicates/aliases, keep order
        names = <map object at 0x000002550757DE40>
     54 super().addItems(list(_names))

File ~\spbstu_projects\napari_viewer\.venv\Lib\site-packages\superqt\combobox\_enum_combobox.py:16, in _get_name(enum_value=<FileFormat.ImageJ TIFF: 1>)
     13 def _get_name(enum_value: Enum):
     14     """Create human readable name if user does not implement `__str__`."""
     15     if (
---> 16         enum_value.__str__.__module__ != "enum"
        enum_value = <FileFormat.ImageJ TIFF: 1>
        enum_value.__str__ = <method-wrapper '__repr__' of FileFormat object at 0x00000254C6B164C0>
     17         and not enum_value.__str__.__module__.startswith("shibokensupport")
     18     ):
     19         # check if function was overloaded
     20         name = str(enum_value)
     21     else:

AttributeError: 'method-wrapper' object has no attribute '__module__'```

tlambert03, 2023-09-13:
this is an error in superqt.  i'll open an issue for you there

closed 2023-09-13 by tlambert03

mshnschnko, 2023-09-13:
@tlambert03, thanks!