pymmcore-plus/napari-micromanager #287: Embedding live image to QT window

mshnschnko, 2023-09-20:
I want to use the live broadcast from the camera from the napari micro manager, but output the picture to the area of my qt window, in the frame. I have already embedded Snap Live ToolBar. Can you tell me where the image is transmitted to display the live image in my window, and not in napari.Viewer (since I don't need the whole viewer window). I know that viewer.window.qt_viewer and viewer.window._qt_window are QWidget objects, but I only need an object or a place to display the image from live.

tlambert03, 2023-09-20:
if you don't need napari, then I'd suggest not using this package, but rather use pymmcore-widgets directly.

There is an example of a simple image preview widget here: https://pymmcore-plus.github.io/pymmcore-widgets/widgets/ImagePreview/
and the source code for that widget is here: https://github.com/pymmcore-plus/pymmcore-widgets/blob/main/src/pymmcore_widgets/_image_widget.py

the key is connecting your widget callbacks to various core events, as done [here](https://github.com/pymmcore-plus/pymmcore-widgets/blob/ba74e3e685899c02baca2ab6200dd4017144a1a7/src/pymmcore_widgets/_image_widget.py#L65-L69)

```python
        ev = self._mmcore.events
        ev.imageSnapped.connect(self._on_image_snapped)
        ev.continuousSequenceAcquisitionStarted.connect(self._on_streaming_start)
        ev.sequenceAcquisitionStopped.connect(self._on_streaming_stop)
        ev.exposureChanged.connect(self._on_exposure_changed)
```

and then handling a new image as desired.  Note that for live acquisition (as opposed to individual snaps), you'll need to start a qtimer that calls `getLastImage()` periodically (as there is no event emitted) as done [here](https://github.com/pymmcore-plus/pymmcore-widgets/blob/ba74e3e685899c02baca2ab6200dd4017144a1a7/src/pymmcore_widgets/_image_widget.py#L94)

mshnschnko, 2023-09-20:
Thank you very much. It really worked for me.

closed 2023-09-20 by mshnschnko