pymmcore-plus/napari-micromanager #288: Error with Nikon Ti

ReddysLab, 2023-09-20:
Hello napari-micromanager Team,

I can control the microscope through the micromanager but unfortunately, I'm getting the below-mentioned error while starting the software the napari either with 'python launch-dev.py' or 'napari -w napari-micromanager'
Could you please help me fix this issue so that I can control the Nikon Ti Microscope using napari-micromanager?

---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
File ~\AppData\Local\Programs\Python\Python311\Lib\site-packages\micromanager_gui\_gui_objects\_objective_widget.py:60, in MMObjectivesWidget._on_sys_cfg_loaded(self=<micromanager_gui._gui_objects._objective_widget.MMObjectivesWidget object>)
     58     self._objective_device = self._guess_objective_device()
     59 self._clear_previous_device_widget()
---> 60 self._combo = self._create_objective_combo(self._objective_device)
        self._objective_device = 'TINosePiece'
        self._combo = <PyQt5.QtWidgets.QComboBox object at 0x000001CEC76B3640>
        self = <micromanager_gui._gui_objects._objective_widget.MMObjectivesWidget object at 0x000001CEC76B3490>
     61 self.layout().addWidget(self._combo)

File ~\AppData\Local\Programs\Python\Python311\Lib\site-packages\micromanager_gui\_gui_objects\_objective_widget.py:86, in MMObjectivesWidget._create_objective_combo(self=<micromanager_gui._gui_objects._objective_widget.MMObjectivesWidget object>, device_label='TINosePiece')
     82 def _create_objective_combo(
     83     self, device_label
     84 ) -> Union[StateDeviceWidget, QComboBox]:
     85     if device_label:
---> 86         combo = _ObjectiveStateWidget(device_label, mmcore=self._mmc)
        device_label = 'TINosePiece'
        _ObjectiveStateWidget = <class 'micromanager_gui._gui_objects._objective_widget._ObjectiveStateWidget'>
        self._mmc = <CMMCorePlus at 0x1cec7652430>
        self = <micromanager_gui._gui_objects._objective_widget.MMObjectivesWidget object at 0x000001CEC76B3490>
     87         combo.setMinimumWidth(285)
     88     else:

File ~\AppData\Local\Programs\Python\Python311\Lib\site-packages\micromanager_gui\_core_widgets\_device_widget.py:124, in StateDeviceWidget.__init__(self=<micromanager_gui._gui_objects._objective_widget._ObjectiveStateWidget object>, device_label='TINosePiece', parent=None, mmcore=<CMMCorePlus>)
    122 self._combo.currentIndexChanged.connect(self._on_combo_changed)
    123 self._refresh_choices()
--> 124 self._combo.setCurrentText(self._mmc.getStateLabel(self._device_label))
        self._combo = <PyQt5.QtWidgets.QComboBox object at 0x000001CEC76EB370>
        self = <micromanager_gui._gui_objects._objective_widget._ObjectiveStateWidget object at 0x000001CEC76EB400>
        self._device_label = 'TINosePiece'
        self._mmc = <CMMCorePlus at 0x1cec7652430>
    126 self.setLayout(QHBoxLayout())
    127 self.layout().setContentsMargins(0, 0, 0, 0)

RuntimeError: Cannot get current position label

tlambert03, 2023-09-20:
That stack trace looks like a very old version... can you check your version of napari micromanager, then update and try again?

ReddysLab, 2023-09-20:
re-installed napari-micromanager 0.0.1 
Still getting the below error

napari -w napari-micromanager
0.00s - Debugger warning: It seems that frozen modules are being used, which may
0.00s - make the debugger miss breakpoints. Please pass -Xfrozen_modules=off
0.00s - to python to disable frozen modules.
0.00s - Note: Debugging will proceed. Set PYDEVD_DISABLE_FILE_VALIDATION=1 to disable this validation.
15:53:12 DEBUG Registering codec 'zlib'
15:53:12 DEBUG Registering codec 'gzip'
15:53:12 DEBUG Registering codec 'bz2'
15:53:12 DEBUG Registering codec 'lzma'
15:53:12 DEBUG Registering codec 'blosc'
15:53:12 DEBUG Registering codec 'zstd'
15:53:12 DEBUG Registering codec 'lz4'
15:53:12 DEBUG Registering codec 'astype'
15:53:12 DEBUG Registering codec 'delta'
15:53:12 DEBUG Registering codec 'quantize'
15:53:12 DEBUG Registering codec 'fixedscaleoffset'
15:53:12 DEBUG Registering codec 'packbits'
15:53:12 DEBUG Registering codec 'categorize'
15:53:12 DEBUG Registering codec 'pickle'
15:53:12 DEBUG Registering codec 'base64'
15:53:12 DEBUG Registering codec 'shuffle'
15:53:12 DEBUG Registering codec 'bitround'
15:53:12 DEBUG Registering codec 'crc32'
15:53:12 DEBUG Registering codec 'adler32'
15:53:12 DEBUG Registering codec 'json2'
15:53:12 DEBUG Registering codec 'vlen-utf8'
15:53:12 DEBUG Registering codec 'vlen-bytes'
15:53:12 DEBUG Registering codec 'vlen-array'
15:53:12 DEBUG Registering codec 'n5_wrapper'
15:53:12 DEBUG Registering processor of napari._qt.qthreading.FunctionWorker[napari.types.LayerDataTuple]: <function _add_worker_data_from_tuple at 0x0000029B9EEB3420> (weight: 0, subclassable: False)
15:53:12 DEBUG Registering processor of napari._qt.qthreading.FunctionWorker[typing.List[napari.types.LayerDataTuple]]: <function _add_worker_data_from_tuple at 0x0000029B9EEB3420> (weight: 0, subclassable: False)
15:53:12 DEBUG Registering processor of napari._qt.qthreading.FunctionWorker[napari.types.ImageData]: <function _add_worker_data at 0x0000029B9EEB3380> (weight: 0, subclassable: False)
15:53:12 DEBUG Registering processor of napari._qt.qthreading.FunctionWorker[napari.types.VectorsData]: <function _add_worker_data at 0x0000029B9EEB3380> (weight: 0, subclassable: False)
15:53:12 DEBUG Registering processor of napari._qt.qthreading.FunctionWorker[napari.types.SurfaceData]: <function _add_worker_data at 0x0000029B9EEB3380> (weight: 0, subclassable: False)
15:53:12 DEBUG Registering processor of napari._qt.qthreading.FunctionWorker[napari.types.PointsData]: <function _add_worker_data at 0x0000029B9EEB3380> (weight: 0, subclassable: False)
15:53:12 DEBUG Registering processor of napari._qt.qthreading.FunctionWorker[napari.types.ShapesData]: <function _add_worker_data at 0x0000029B9EEB3380> (weight: 0, subclassable: False)
15:53:12 DEBUG Registering processor of napari._qt.qthreading.FunctionWorker[napari.types.TracksData]: <function _add_worker_data at 0x0000029B9EEB3380> (weight: 0, subclassable: False)
15:53:12 DEBUG Registering processor of napari._qt.qthreading.FunctionWorker[napari.types.LabelsData]: <function _add_worker_data at 0x0000029B9EEB3380> (weight: 0, subclassable: False)
15:53:13 DEBUG Registering provider of <class 'napari._qt.qt_main_window.Window'>: <function init_qactions.<locals>._provide_window at 0x0000029BAD3AB880> (weight: 0, subclassable: True)
15:53:13 DEBUG Registering provider of <class 'napari._qt.qt_viewer.QtViewer'>: <function init_qactions.<locals>._provide_qt_viewer at 0x0000029BAD3ABD80> (weight: 0, subclassable: True)
15:53:13 DEBUG Executing @injected ViewerToggleAction.__init__.<locals>.get_current(viewer: napari.viewer.Viewer) with args: (), kwargs: {}
15:53:13 DEBUG Rebuilding provider map cache
15:53:13 DEBUG   injecting viewer: <class 'napari.viewer.Viewer'> = Viewer(camera=Camera(center=(0.0, 0.0, 0.0), zoom=1.0, angles=(0.0, 0.0, 90.0), perspective=0.0, mouse_pan=True, mouse_zoom=True), cursor=Cursor(position=(1.0, 1.0), scaled=True, size=1, style=<CursorStyle.STANDARD: 'standard'>), dims=Dims(ndim=2, ndisplay=2, last_used=0, range=((0, 2, 1), (0, 2, 1)), current_step=(0, 0), order=(0, 1), axis_labels=('0', '1')), grid=GridCanvas(stride=1, shape=(-1, -1), enabled=False), layers=[], help='', status='Ready', tooltip=Tooltip(visible=False, text=''), theme='dark', title='napari', mouse_over_canvas=False, mouse_move_callbacks=[], mouse_drag_callbacks=[], mouse_double_click_callbacks=[], mouse_wheel_callbacks=[<function dims_scroll at 0x0000029B9DB1FC40>], _persisted_mouse_event={}, _mouse_drag_gen={}, _mouse_wheel_gen={}, keymap={})
15:53:13 DEBUG   Calling ViewerToggleAction.__init__.<locals>.get_current with {'viewer': Viewer(camera=Camera(center=(0.0, 0.0, 0.0), zoom=1.0, angles=(0.0, 0.0, 90.0), perspective=0.0, mouse_pan=True, mouse_zoom=True), cursor=Cursor(position=(1.0, 1.0), scaled=True, size=1, style=<CursorStyle.STANDARD: 'standard'>), dims=Dims(ndim=2, ndisplay=2, last_used=0, range=((0, 2, 1), (0, 2, 1)), current_step=(0, 0), order=(0, 1), axis_labels=('0', '1')), grid=GridCanvas(stride=1, shape=(-1, -1), enabled=False), layers=[], help='', status='Ready', tooltip=Tooltip(visible=False, text=''), theme='dark', title='napari', mouse_over_canvas=False, mouse_move_callbacks=[], mouse_drag_callbacks=[], mouse_double_click_callbacks=[], mouse_wheel_callbacks=[<function dims_scroll at 0x0000029B9DB1FC40>], _persisted_mouse_event={}, _mouse_drag_gen={}, _mouse_wheel_gen={}, keymap={})} (injected {'viewer'})
[... cut ...]
        self._wdg = <pymmcore_widgets._objective_widget.ObjectivesWidget object at 0x0000029BACA8C1F0>
        self = <napari_micromanager._gui_objects._toolbar.ObjectivesToolBar object at 0x0000029BACA67F40>
        QSizePolicy.Policy = <class 'PyQt5.QtWidgets.QSizePolicy.Policy'>
        QSizePolicy = <class 'PyQt5.QtWidgets.QSizePolicy'>
    243         QSizePolicy.Policy.Fixed, QSizePolicy.Policy.Fixed
    244     )
    245     self._wdg.setMinimumWidth(0)
    246     self._wdg._combo.adjustSize()

RuntimeError: wrapped C/C++ object of type QComboBox has been deleted

tlambert03, 2023-09-20:
ok, thanks, that's a different error, but one we can look into.  cc @fdrgsp, can you see if you can figure out what might be causing `_resize_obj` to be called after `self._wdg._combo` has been deleted?

tlambert03, 2023-09-21:
@ReddysLab, would you mind letting us know what operating system and python version you have?  Also, if you don't mind copying your `config.cfg` file here it would be helpful.  @fdrgsp thinks he has a fix, but we're unable to reproduce it here as of yet.  thanks!

tlambert03, 2023-09-21:
Actually, I just merged #290, which should fix this issue for you either way.  Just pushed v0.0.2 and it will be available on pypi shortly.  curious to hear if it works.  (please do let us know the above info when you get a chance though!)

ReddysLab, 2023-09-22:
Many thanks for kindly releasing an update. Unfortunately, this hasn't fixed the issue.
I did try three methods to launch the program. 'python launch-dev.py'.  'python launch-dev.py -Xfrozen_modules=off'     'napari -w napari-micromanager' followed by loading mm config file. None of them worked and gave below-mentioned errors. Could you please help me to fix these issues?

Windows 10 Pro 64-bit 
Python 3.11.5
pip list
Package                       Version
----------------------------- -----------
alabaster                     0.7.13
app-model                     0.2.1
appdirs                       1.4.4
asciitree                     0.3.3
asttokens                     2.2.1
attrs                         23.1.0
Babel                         2.12.1
backcall                      0.2.0
build                         1.0.0
cachey                        0.2.1
certifi                       2023.7.22
charset-normalizer            3.2.0
click                         8.1.7
cloudpickle                   2.2.1
colorama                      0.4.6
comm                          0.1.4
dask                          2023.9.0
debugpy                       1.6.7.post1
decorator                     5.1.1
docstring-parser              0.15
docutils                      0.17.1
entrypoints                   0.4
executing                     1.2.0
fasteners                     0.19
fonticon-materialdesignicons6 6.9.96
freetype-py                   2.4.0
fsspec                        2023.6.0
HeapDict                      1.0.1
hsluv                         5.0.3
idna                          3.4
imageio                       2.31.2
imagesize                     1.4.1
importlib-metadata            6.8.0
in-n-out                      0.1.8
ipykernel                     6.25.1
ipython                       8.15.0
ipython-genutils              0.2.0
jedi                          0.19.0
Jinja2                        3.1.2
jsonschema                    4.19.0
jsonschema-specifications     2023.7.1
jupyter_client                8.3.1
jupyter_core                  5.3.1
kiwisolver                    1.4.5
lazy_loader                   0.3
locket                        1.0.0
magicgui                      0.7.3
markdown-it-py                3.0.0
MarkupSafe                    2.1.3
matplotlib-inline             0.1.6
mdurl                         0.1.2
mypy-extensions               1.0.0
napari                        0.4.18
napari-console                0.0.8
napari-micromanager           0.0.2
napari-plugin-engine          0.2.0
napari-plugin-manager         0.1.0a2
napari-svg                    0.1.10
nest-asyncio                  1.5.7
networkx                      3.1
npe2                          0.7.2
numcodecs                     0.11.0
[... 161 lines cut ...]
        self = <pymmcore_widgets._objective_widget.ObjectivesWidget object at 0x000001FCFEE7C1F0>
        self._mmc = <CMMCorePlus at 0x1fcff819670>
        StateDeviceWidget = <class 'pymmcore_widgets._device_widget.StateDeviceWidget'>
     82         )
     83         combo._combo.currentIndexChanged.connect(self._on_obj_changed)
     84     else:

File ~\AppData\Local\Programs\Python\Python311\Lib\site-packages\pymmcore_widgets\_device_widget.py:141, in StateDeviceWidget.__init__(self=<pymmcore_widgets._device_widget.StateDeviceWidget object>, device_label='TINosePiece', parent=<pymmcore_widgets._objective_widget.ObjectivesWidget object>, mmcore=<CMMCorePlus>)
    139 self._combo.currentIndexChanged.connect(self._on_combo_changed)
    140 self._refresh_choices()
--> 141 self._combo.setCurrentText(self._mmc.getStateLabel(self._device_label))
        self._combo = <PyQt5.QtWidgets.QComboBox object at 0x000001FCFEE7F130>
        self = <pymmcore_widgets._device_widget.StateDeviceWidget object at 0x000001FCFEE7F1C0>
        self._device_label = 'TINosePiece'
        self._mmc = <CMMCorePlus at 0x1fcff819670>
    143 self.setLayout(QHBoxLayout())
    144 self.layout().setContentsMargins(0, 0, 0, 0)

RuntimeError: Cannot get current position label

D:\>

MMConfig.cfg is renamed to MMConfig.txt for attaching purpose

[MMConfig.txt](https://github.com/pymmcore-plus/napari-micromanager/files/12702989/MMConfig.txt)

tlambert03, 2023-09-22:
thanks so much!  

That's very helpful.  sorry for the exceptions, we'll get them ironed out for you shortly :) your config and the traceback are super helpful.  will ping you when it's time to try again

fdrgsp, 2023-09-22:
@ReddysLab , do you mind loading this minimal `Ti` configuration to see if `napari-micromanager` is loading correctly?

Thanks!

```text
# Generated by Configurator on Tue Jul 27 15:50:11 EDT 2023

# Reset
Property,Core,Initialize,0

# Devices
Device,TIScope,NikonTI,TIScope
Device,TIAnalyzer,NikonTI,TIAnalyzer
Device,TIDiaLamp,NikonTI,TIDiaLamp
Device,TINosePiece,NikonTI,TINosePiece
Device,TIFilterBlock1,NikonTI,TIFilterBlock1
Device,TILightPath,NikonTI,TILightPath
Device,TIZDrive,NikonTI,TIZDrive
Device,TIPFSOffset,NikonTI,TIPFSOffset
Device,TIPFSStatus,NikonTI,TIPFSStatus

# Pre-init settings for devices

# Pre-init settings for COM ports

# Hub (parent) references
Parent,TIAnalyzer,TIScope
Parent,TIDiaLamp,TIScope
Parent,TINosePiece,TIScope
Parent,TIFilterBlock1,TIScope
Parent,TILightPath,TIScope
Parent,TIZDrive,TIScope
Parent,TIPFSOffset,TIScope
Parent,TIPFSStatus,TIScope

# Initialize
Property,Core,Initialize,1

# Delays

# Focus directions
FocusDirection,TIZDrive,0
FocusDirection,TIPFSOffset,0

# Roles
#Property,Core,Focus,TIZDrive

# Camera-synchronized devices

# Labels
# TINosePiece
Label,TINosePiece,5,6-Plan Fluor 40x NA 0.75 Dry
Label,TINosePiece,4,5-Unknown
Label,TINosePiece,3,4-Plan Apo VC 60x NA 1.40 Oil
Label,TINosePiece,2,3-Plan Apo 40x NA 0.95 Dry
Label,TINosePiece,1,2-Plan Apo VC 20x NA 0.75 Dry
Label,TINosePiece,0,1-Plan Fluor 10x NA 0.30 Dry
# TIFilterBlock1
Label,TIFilterBlock1,5,6-ANALY
Label,TIFilterBlock1,4,5-HyPer
Label,TIFilterBlock1,3,4-49008
Label,TIFilterBlock1,2,3-49003
Label,TIFilterBlock1,1,2-CFP-S
Label,TIFilterBlock1,0,1-49000
# TILightPath
Label,TILightPath,3,4-Left80
Label,TILightPath,2,3-Right100
Label,TILightPath,1,2-Left100
Label,TILightPath,0,1-Eye100

# Configuration presets

# PixelSize settings

```

ReddysLab, 2023-09-22:
The above config is loading fine and I tested the TIDiaLamp On/Off toggling over gui, and it works.

fdrgsp, 2023-09-22:
> The above config is loading fine and I tested the TIDiaLamp On/Off toggling over gui, and it works.

@ReddysLab That's a good start! 

As a further test, could you use your `MMConfig.cfg` but remove the `TINosePiece` from both the `BrightField` and `LightOFF` presets and load the it again?

e.g. remove:
```
ConfigGroup,SRRF,BrightField,TINosePiece,ExtraDelayMs,20.0000
ConfigGroup,SRRF,BrightField,TINosePiece,Label,6-Plan Fluor 40x NA 0.75 Dry
ConfigGroup,SRRF,BrightField,TINosePiece,State,5
```

ReddysLab, 2023-09-22:
unfortunately, gives the below-mentioned error

17:18:45 DEBUG Unloading all devices
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
File ~\AppData\Local\Programs\Python\Python311\Lib\site-packages\pymmcore_widgets\_objective_widget.py:71, in ObjectivesWidget._on_sys_cfg_loaded(self=<pymmcore_widgets._objective_widget.ObjectivesWidget object>)
     69     self._objective_device = guess_objective_or_prompt(parent=self)
     70 self._combo.setParent(QWidget())
---> 71 self._combo = self._create_objective_combo(self._objective_device)
        self._objective_device = 'TINosePiece'
        self._combo = <PyQt5.QtWidgets.QComboBox object at 0x0000020B03C84280>
        self = <pymmcore_widgets._objective_widget.ObjectivesWidget object at 0x0000020B03C841F0>
     72 self.layout().addWidget(self._combo)

File ~\AppData\Local\Programs\Python\Python311\Lib\site-packages\pymmcore_widgets\_objective_widget.py:81, in ObjectivesWidget._create_objective_combo(self=<pymmcore_widgets._objective_widget.ObjectivesWidget object>, device_label='TINosePiece')
     74 def _create_objective_combo(
     75     self, device_label: str | None
     76 ) -> StateDeviceWidget | QComboBox:
     77     if device_label:
     78         combo = (
     79             _ObjectiveStateWidget(device_label, parent=self, mmcore=self._mmc)
     80             if self._mmc.getFocusDevice()
---> 81             else StateDeviceWidget(device_label, parent=self, mmcore=self._mmc)
        device_label = 'TINosePiece'
        _ObjectiveStateWidget = <class 'pymmcore_widgets._objective_widget._ObjectiveStateWidget'>
        self = <pymmcore_widgets._objective_widget.ObjectivesWidget object at 0x0000020B03C841F0>
        self._mmc = <CMMCorePlus at 0x20b03e41580>
        StateDeviceWidget = <class 'pymmcore_widgets._device_widget.StateDeviceWidget'>
     82         )
     83         combo._combo.currentIndexChanged.connect(self._on_obj_changed)
     84     else:

File ~\AppData\Local\Programs\Python\Python311\Lib\site-packages\pymmcore_widgets\_device_widget.py:141, in StateDeviceWidget.__init__(self=<pymmcore_widgets._device_widget.StateDeviceWidget object>, device_label='TINosePiece', parent=<pymmcore_widgets._objective_widget.ObjectivesWidget object>, mmcore=<CMMCorePlus>)
    139 self._combo.currentIndexChanged.connect(self._on_combo_changed)
    140 self._refresh_choices()
--> 141 self._combo.setCurrentText(self._mmc.getStateLabel(self._device_label))
        self._combo = <PyQt5.QtWidgets.QComboBox object at 0x0000020B03C87130>
        self = <pymmcore_widgets._device_widget.StateDeviceWidget object at 0x0000020B03C871C0>
        self._device_label = 'TINosePiece'
        self._mmc = <CMMCorePlus at 0x20b03e41580>
    143 self.setLayout(QHBoxLayout())
    144 self.layout().setContentsMargins(0, 0, 0, 0)

RuntimeError: Cannot get current position label

D:\>

ReddysLab, 2023-09-22:
it seems like

Device,OkoLab,Okolab,OkoLab

is causing some issues. If I remove the related lines from the config file.  napari -w napari-micromanager  loads okay and there are some other issues

# Generated by Configurator on Tue Jul 27 15:50:11 EDT 2023

# Reset
Property,Core,Initialize,0

# Devices
Device,COM6,SerialManager,COM6
Device,COM3,SerialManager,COM3
Device,Andor,Andor,Andor
Device,TIScope,NikonTI,TIScope
Device,TIAnalyzer,NikonTI,TIAnalyzer
Device,TIEpiShutter,NikonTI,TIEpiShutter
Device,TIDiaShutter,NikonTI,TIDiaShutter
Device,TIDiaLamp,NikonTI,TIDiaLamp
Device,TINosePiece,NikonTI,TINosePiece
Device,TICondenserCassette,NikonTI,TICondenserCassette
Device,TIFilterBlock1,NikonTI,TIFilterBlock1
Device,TILightPath,NikonTI,TILightPath
Device,TIZDrive,NikonTI,TIZDrive
Device,TIXYDrive,NikonTI,TIXYDrive
Device,TIPFSOffset,NikonTI,TIPFSOffset
Device,TIPFSStatus,NikonTI,TIPFSStatus
Device,pE4000,pE4000,pE4000
Device,Shutter-A,SutterLambda,Shutter-A

# Pre-init settings for devices
Property,COM6,AnswerTimeout,500.0000
Property,COM6,BaudRate,128000
Property,COM6,DTR,Disable
Property,COM6,DataBits,8
Property,COM6,DelayBetweenCharsMs,0.0000
Property,COM6,Fast USB to Serial,Disable
Property,COM6,Handshaking,Off
Property,COM6,Parity,None
Property,COM6,StopBits,1
Property,COM6,Verbose,1
Property,COM3,AnswerTimeout,500.0000
Property,COM3,BaudRate,9600
Property,COM3,DTR,Disable
Property,COM3,DataBits,8
Property,COM3,DelayBetweenCharsMs,0.0000
Property,COM3,Fast USB to Serial,Disable
Property,COM3,Handshaking,Off
Property,COM3,Parity,None
Property,COM3,StopBits,1
Property,COM3,Verbose,1
Property,Andor,Camera,| iXon Ultra | DU888_BV | 13420 |
Property,pE4000,Port,COM3
Property,Shutter-A,Port,COM6

# Pre-init settings for COM ports

# Hub (parent) references
Parent,TIAnalyzer,TIScope
Parent,TIEpiShutter,TIScope
Parent,TIDiaShutter,TIScope
Parent,TIDiaLamp,TIScope
Parent,TINosePiece,TIScope
Parent,TICondenserCassette,TIScope
Parent,TIFilterBlock1,TIScope
Parent,TILightPath,TIScope
Parent,TIZDrive,TIScope
Parent,TIXYDrive,TIScope
Parent,TIPFSOffset,TIScope
Parent,TIPFSStatus,TIScope

# Initialize
Property,Core,Initialize,1

# Delays
Delay,TIEpiShutter,20.0
Delay,TIDiaShutter,20.0
Delay,TIDiaLamp,20.0
Delay,TINosePiece,20.0
Delay,TICondenserCassette,20.0
Delay,TIFilterBlock1,20.0
Delay,TILightPath,20.0
Delay,pE4000,20.0
Delay,Shutter-A,20.0

# Focus directions
FocusDirection,TIZDrive,0
FocusDirection,TIPFSOffset,0

# Roles
Property,Core,Camera,Andor
Property,Core,AutoShutter,1

# Camera-synchronized devices

# Labels
# TINosePiece
Label,TINosePiece,5,6-Plan Fluor 40x NA 0.75 Dry
Label,TINosePiece,4,5-Unknown
Label,TINosePiece,3,4-Plan Apo VC 60x NA 1.40 Oil
Label,TINosePiece,2,3-Plan Apo 40x NA 0.95 Dry
Label,TINosePiece,1,2-Plan Apo VC 20x NA 0.75 Dry
Label,TINosePiece,0,1-Plan Fluor 10x NA 0.30 Dry
# TICondenserCassette
Label,TICondenserCassette,4,5------
Label,TICondenserCassette,3,4------
Label,TICondenserCassette,2,3------
Label,TICondenserCassette,1,2-DICN2
Label,TICondenserCassette,0,1-A
# TIFilterBlock1
Label,TIFilterBlock1,5,6-ANALY
Label,TIFilterBlock1,4,5-HyPer
Label,TIFilterBlock1,3,4-49008
Label,TIFilterBlock1,2,3-49003
Label,TIFilterBlock1,1,2-CFP-S
Label,TIFilterBlock1,0,1-49000
# TILightPath
Label,TILightPath,3,4-Left80
Label,TILightPath,2,3-Right100
Label,TILightPath,1,2-Left100
Label,TILightPath,0,1-Eye100

# Configuration presets

# PixelSize settings

ReddysLab, 2023-09-22:
I cannot toggle On/Off the PE4000 channels (A, B, C, D) independently through software. This requires manually clicking the button on the device pad. Micromanager can toggle the 4 channels independently.

[image]

fdrgsp, 2023-09-22:
> it seems like Device,OkoLab,Okolab,OkoLab is causing some issues. If I remove the related lines from the config file. napari -w napari-micromanager loads okay and there are some other issues

@ReddysLab So basically if you remove the `OkoLab` device you are not getting any error from the `TINosePiece` and everything works as expected apart from the PE4000?

> Micromanager can toggle the 4 channels independently

Do you mean that you can see the 4 different PE4000 channels in the Micromanager  `Shutter` drop-down menu (as in the image below)?

[image]

or you can select them from the `Group/Preset` table (image below)?

[image]

ReddysLab, 2023-09-29:
You are right, after removing the OkoLab device I'm not getting any error from the TINosePiece.
I'm still configuring the Group/Preset to test the software thoroughly for our purpose. Sometimes randomly napari GUI crashes while configuring/tweaking the intensity of the light source. I will get back to you if it happens again. Many thanks for your kind help.

tlambert03, 2024-01-26:
closing, but feel free to reopen if you have any more issues

PR #290 (2023-09-21): fix: ObjectivesToolBar
This is related to #288 

merge after https://github.com/pymmcore-plus/pymmcore-widgets/pull/196

In this PR we are removing the `ObjectivesToolBar` `_resize_obj()` method and we moved it directly to `pymmocre-widgets`  `ObjectivesWidget()` (https://github.com/pymmcore-plus/pymmcore-widgets/pull/196). This should fix the issue found in #288  since resizing is performed directly when the combo is created.

closes #288.
closed 2024-01-26 by tlambert03