pymmcore-plus/napari-micromanager #289: Feature request - Add "Keep shutter open" option for z-stack

GabeK-L, 2023-09-20:
Hello, 

I have been using napari-micromanager to control an IX-81 with CSU-X1 spinning disk. Would it be possible to add in the "Keep shutter open" check box that is available in micromanager so that I can take fast z-stacks? Closing the shutter between each z-acquisition dramatically slows down my system; I have my piezo stage triggered by my camera but I cannot currently take advantage of this if the shutter is closed between each step. Ideally, the shutter would only close between channels, positions, etc.

Thanks!
Gabe

tlambert03, 2023-09-20:
yep, it's been added to a new-and-improved MDA widget in https://github.com/pymmcore-plus/pymmcore-widgets/pull/180 

that widget hasn't made it's way into napari-micromanager yet. but will soon..

GabeK-L, 2023-09-20:
Awesome! Thanks for the quick reply and the info. I will give the MDA widget a try and will keep an eye out for that update.