pymmcore-plus/napari-micromanager #295: MDA not working when using 12 bit images (Photometric Kinetix in Sensitivity mode).

raulbola, 2023-10-13:
Hi,

I have been testing different acquisitions using the MDA widget and for some cases I didn't manage to suceed. Basically I start from the test configuration file, delete the demo camera and add my camera setup (Photometrics Kinetix).

The camera works perfectly fine in Live preview mode and I can change the different parameters, however when trying to acquire multiple images with MDA i found two different issues. The first issue has to do with the "Sensitivity" mode of Kinetix, the camera is working in 12 bit mode and the MDA does not change it to 16bit (as Micromanager does). However it is strange because the Live Preview mode works flawlessly.

The error:
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
File ~\anaconda3\envs\holoAOD\Lib\site-packages\superqt\utils\_ensure_thread.py:97, in ensure_main_thread.<locals>._out_func.<locals>._func(_max_args_=2, *args=(<napari_micromanager._mda_handler._NapariMDAHandler object>, MDASequence(metadata={'napari_mm_sequence_meta':...ime.timedelta(microseconds=200000), loops=200)]))), **kwargs={})
     95 @wraps(func_)
     96 def _func(*args, _max_args_=max_args, **kwargs):
---> 97     return _run_in_thread(
        func_ = <function _NapariMDAHandler._on_mda_started at 0x0000016FD20B5A80>
        QCoreApplication = <class 'PyQt5.QtCore.QCoreApplication'>
        await_return = False
        timeout = 1000
        args[:_max_args_] = (<napari_micromanager._mda_handler._NapariMDAHandler object at 0x0000016FDD708110>, MDASequence(metadata={'napari_mm_sequence_meta': SequenceMeta(mode='mda', split_channels=False, should_save=False, file_name='Experiment', save_dir='', save_pos=False)}, stage_positions=(Position(x=-0.0, y=-0.0, z=0.0, name='Pos000', sequence=None),), channels=(Channel(config='Camera-left', group='LightPath', exposure=10.0, do_stack=True, z_offset=0.0, acquire_every=1, camera=None),), time_plan=MultiPhaseTimePlan(prioritize_duration=False, phases=[TIntervalLoops(prioritize_duration=False, interval=datetime.timedelta(microseconds=200000), loops=200)])))
        args = (<napari_micromanager._mda_handler._NapariMDAHandler object at 0x0000016FDD708110>, MDASequence(metadata={'napari_mm_sequence_meta': SequenceMeta(mode='mda', split_channels=False, should_save=False, file_name='Experiment', save_dir='', save_pos=False)}, stage_positions=(Position(x=-0.0, y=-0.0, z=0.0, name='Pos000', sequence=None),), channels=(Channel(config='Camera-left', group='LightPath', exposure=10.0, do_stack=True, z_offset=0.0, acquire_every=1, camera=None),), time_plan=MultiPhaseTimePlan(prioritize_duration=False, phases=[TIntervalLoops(prioritize_duration=False, interval=datetime.timedelta(microseconds=200000), loops=200)])))
        _max_args_ = 2
        kwargs = {}
     98         func_,
     99         QCoreApplication.instance().thread(),
    100         await_return,
    101         timeout,
    102         args[:_max_args_],
    103         kwargs,
    104     )

File ~\anaconda3\envs\holoAOD\Lib\site-packages\superqt\utils\_ensure_thread.py:181, in _run_in_thread(func=<function _NapariMDAHandler._on_mda_started>, thread=<PyQt5.QtCore.QThread object>, await_return=False, timeout=1000, args=(<napari_micromanager._mda_handler._NapariMDAHandler object>, MDASequence(metadata={'napari_mm_sequence_meta':...ime.timedelta(microseconds=200000), loops=200)]))), kwargs={})
    179 future = Future()  # type: ignore
    180 if thread is QThread.currentThread():
--> 181     result = func(*args, **kwargs)
        func = <function _NapariMDAHandler._on_mda_started at 0x0000016FD20B5A80>
        args = (<napari_micromanager._mda_handler._NapariMDAHandler object at 0x0000016FDD708110>, MDASequence(metadata={'napari_mm_sequence_meta': SequenceMeta(mode='mda', split_channels=False, should_save=False, file_name='Experiment', save_dir='', save_pos=False)}, stage_positions=(Position(x=-0.0, y=-0.0, z=0.0, name='Pos000', sequence=None),), channels=(Channel(config='Camera-left', group='LightPath', exposure=10.0, do_stack=True, z_offset=0.0, acquire_every=1, camera=None),), time_plan=MultiPhaseTimePlan(prioritize_duration=False, phases=[TIntervalLoops(prioritize_duration=False, interval=datetime.timedelta(microseconds=200000), loops=200)])))
        kwargs = {}
    182     if not await_return:
    183         future.set_result(result)

File ~\anaconda3\envs\holoAOD\Lib\site-packages\napari_micromanager\_mda_handler.py:123, in _NapariMDAHandler._on_mda_started(self=<napari_micromanager._mda_handler._NapariMDAHandler object>, sequence=MDASequence(metadata={'napari_mm_sequence_meta':...ime.timedelta(microseconds=200000), loops=200)])))
    121 dtype = f"uint{self._mmc.getImageBitDepth()}"
    122 # create the zarr array and add it to the viewer
--> 123 z = zarr.open(
        tmp = <TemporaryDirectory 'C:\\Users\\CRESTO~1\\AppData\\Local\\Temp\\tmpqjflqqup'>
        str(tmp.name) = 'C:\\Users\\CRESTO~1\\AppData\\Local\\Temp\\tmpqjflqqup'
        tmp.name = 'C:\\Users\\CRESTO~1\\AppData\\Local\\Temp\\tmpqjflqqup'
        yx_shape = [3200, 3200]
        shape = [200, 1, 1]
        shape + yx_shape = [200, 1, 1, 3200, 3200]
        dtype = 'uint12'
        tuple([1] * len(shape) + yx_shape) = (1, 1, 1, 3200, 3200)
        [1] * len(shape) + yx_shape = [1, 1, 1, 3200, 3200]
        [1] * len(shape) = [1, 1, 1]
        len(shape) = 3
    124     str(tmp.name),
    125     shape=shape + yx_shape,
    126     dtype=dtype,
    127     chunks=tuple([1] * len(shape) + yx_shape),  # VERY IMPORTANT FOR SPEED!
    128 )
    129 fname = meta.file_name if meta.should_save else "Exp"
    130 self._create_empty_image_layer(z, f"{fname}_{id_}", sequence, **kwargs)

File ~\anaconda3\envs\holoAOD\Lib\site-packages\zarr\convenience.py:120, in open(store=r'C:\Users\CRESTO~1\AppData\Local\Temp\tmpqjflqqup', mode='a', zarr_version=None, path='', **kwargs={'chunks': (1, 1, 1, 3200, 3200), 'dtype': 'uint12', 'path': '', 'shape': [200, 1, 1, 3200, 3200]})
    118 elif mode == "a":
    119     if "shape" in kwargs or contains_array(_store, path):
--> 120         return open_array(_store, mode=mode, **kwargs)
        _store = <zarr.storage.DirectoryStore object at 0x0000016FDF330490>
        kwargs = {'shape': [200, 1, 1, 3200, 3200], 'dtype': 'uint12', 'chunks': (1, 1, 1, 3200, 3200), 'path': ''}
        mode = 'a'
    121     else:
[... 10 lines cut ...]
    617 if mode in ["r", "r+"]:

TypeError: data type 'uint12' not understood

__________________________________________________________________________________________________________________________

Then, once you switch to another camera mode, for example you change to 16bit mode and change back to 12bit, the acquisition works but the error you get is:
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
File ~\anaconda3\envs\holoAOD\Lib\site-packages\pymmcore_widgets\_property_widget.py:344, in PropertyWidget._on_core_change(self=<pymmcore_widgets._property_widget.PropertyWidget object>, dev_label='Shutter', prop_name='State', new_val=(False,))
    342 if dev_label == self._device_label and prop_name == self._prop_name:
    343     with utils.signals_blocked(self._value_widget):
--> 344         self._value_widget.setValue(new_val)
        self = <pymmcore_widgets._property_widget.PropertyWidget object at 0x00000218D2F45E10>
        self._value_widget = <pymmcore_widgets._property_widget.IntBoolWidget object at 0x00000218D2F45FC0>
        new_val = (False,)

File ~\anaconda3\envs\holoAOD\Lib\site-packages\pymmcore_widgets\_property_widget.py:127, in IntBoolWidget.setValue(self=<pymmcore_widgets._property_widget.IntBoolWidget object>, val=(False,))
    125 def setValue(self, val: str | int) -> None:
    126     """Set value."""
--> 127     return self.setChecked(bool(int(val)))
        self = <pymmcore_widgets._property_widget.IntBoolWidget object at 0x00000218D2F45FC0>
        val = (False,)

TypeError: int() argument must be a string, a bytes-like object or a real number, not 'tuple'

tlambert03, 2023-10-13:
agreed, many thanks for the issues 👍

a quick tip!  when pasting code into a github issue, if you wrap the code in three backticks:

````md
```
some code
```
````

it will format it nicely :)  I updated your post above

raulbola, 2023-10-13:
> agreed, many thanks for the issues 👍
> 
> a quick tip! when pasting code into a github issue, if you wrap the code in three backticks:
> 
> ```
> ```
> some code
> ```
> ```
> 
> it will format it nicely :) I updated your post above

Hi @tlambert03 , thank you very much for the tip 👍 . I'm relatively new to github and would like to apologize for any bad practices.

tlambert03, 2023-10-13:
no apology needed :)

raulbola, 2023-10-13:
> Hi @raulbola, thanks!
> 
> I think the first issue happens when we create the zarr array and is due to the "uint12" type not existing, we will fix it soon!
> 
> About the second issue, I believe we fixed yesterday in `pymmcore-plus` but we did not release the version with the new fix yet. We'll do that soon!
> 
> Thanks again for these helpful notes!

Hi @fdrgsp , yes the uint12 pixel type is not very common. For me it was strange because Micromanager already casts the image from uint12 to uint16 when using this mode in the Kinetix camera. I am glad that maybe you will include uint12 in the zarr array. However, I think that this is a minor issue, I can just switch to Dynamic Range (16bit) mode.

Regarding the second error, I am getting it everytime I run an acquisition. Do you think is a configuration error?

fdrgsp, 2023-10-13:
> Regarding the second error, I am getting it everytime I run an acquisition. Do you think is a configuration error?

Hi @raulbola! no I don't think it is a configuration error. It's a bug `we` discover yesterday on `pymmcore-plus` related to one of the core shutter method. Your .cfg should be ok 👍🏻!

PR #307 (2024-01-27): fix: fix zarr dtype
To set the zarr dtype, use `getBytesPerPixel`  instead of  `getImageBitDepth` but get this info from using `getBytesPerPixel` 

closes #295
closed 2024-01-27 by tlambert03