pymmcore-plus/napari-micromanager #296: How to connect to ASI Stage on Linux

nkhmicrographia, 2023-10-17:
Hi! Great job on napari-micromanager project!

I am trying to connect an ASI Stage to napari-micromanager on Linux (Ubuntu 22.04), my micromanager mmCoreAndDevices build has the ASI .so files, but when I connect the stage and run naoari-micromanager, I don't see it in the device browser, any specific tricks I am missing here? 

The stage shows up under `/dev/ttyUSB0` when I connect it so the serial/usb connection seems to be working. 

Any guidance would be highly appreciated!

tlambert03, 2023-10-17:
is this specific to napari-micromanager?  that is, does it appear when you load the same config file in mm-studio (the java based micro-manager app)?

If it works in the java app but not in napari-micromanager, can you share your config file please?

nkhmicrographia, 2023-10-17:
> does it appear when you load the same config file in mm-studio

Thanks @tlambert03 , that was a very helpful comment! I generated the config file using micro-manager and used it, works perfectly fine, it was more me expecting the device property browser to search for my connected devices without a config file - which I see now is clearly a wrong expectation. 

Does napari-micromanager provide anything equivalent to micromanager's hardware configuration wizard GUI (or alternative CLI interface to generate a config file)? At the moment one still needs micromanager to generate the configs.

closed 2023-10-17 by nkhmicrographia