pymmcore-plus/napari-micromanager #303: Hardware autofocus implementation

GabeK-L, 2024-01-11:
Hi,

I have been testing napari-micromanager with Olympus IX81 microscopes. Are there any plans to add hardware autofocus into the MDA like there is in micromanager?

Thanks!

tlambert03, 2024-01-11:
Hey @GabeK-L,
pymmcore-plus and useq-schema do support hardware autofocus, and the latest version of the MDA widget from pymmcore-widgets should have it in there... do you see something like this? (if not, update pymmcore-widgets)

[image]

the way to interpret those checkboxes is that hardware autofocus will be performed whenever the index of the corresponding axis changes.  So, for example, if you check the "p" box, autofocus will be performed each time you move to a new position, if you check "t", it will be performed at each time point, etc...

let me know how that goes for you

tlambert03, 2024-01-11:
will also mention that most of our testing has been on Nikon PFS, so please do let us know if you run into any issues with the IX-81

GabeK-L, 2024-01-12:
Awesome, thanks for the quick response. I will test it out and let you know if I have any issues.

tlambert03, 2024-01-26:
I'm going to close this, but definitely do comment if you try it and have any issues

closed 2024-01-26 by tlambert03