pymmcore-plus/napari-micromanager #305: question: using napari termial, how can I stream the newly acquired images to the viewer?

linshaova, 2024-01-26:
Hello,

After launching napari-MM and verifying that MM configuration and camera etc. worked, I opened its Python console and obtained a reference `mmc` to the same core object using the `CMMCorePlus.instance` call (verified with `mmc.getFocusDevice` for example). However, when I then ran `mmc.snapImage` or `mmc.run_mda`, I didn't see the default image viewer updating with the newly acquired images. What other steps are needed? Thanks!

-lin

tlambert03, 2024-01-26:
- `mmc.snapImage`

This one doesn't immediately show anything is because (perhaps surprisingly) it doesn't emit any event that napari is listening to;  `mmc.snap()` will however (it's a combination of `snapImage()` and `getImage()`, added in pymmcore-plus)
this confusion is also discussed in: https://github.com/pymmcore-plus/pymmcore-plus/issues/144

- `mmc.run_mda`

This one doesn't work at the moment because, currently, napari-micromanager only pays attention to MDAs that it started itself.  This is a bit of an arbitrary decision, and probably overly "cautious". 

It's caused by this line right here:

https://github.com/pymmcore-plus/napari-micromanager/blob/ebddc0aee75d85011ac07ff3365d513c9abd28e4/src/napari_micromanager/_mda_handler.py#L194C1-L196C30

it would be *very* easy to change that behavior.  If you want to "hack" it to just work for you, all you need to do is add a little metadata to the MDASequence:

```python
from useq import MDASequence
from napari_micromanager._mda_meta import SequenceMeta

fake_meta = {'napari_mm_sequence_meta': SequenceMeta()}
core.run_mda(MDASequence(time_plan={'loops': 5, 'interval': 0.1}, metadata= fake_meta))
```

what do we think, lin and @fdrgsp, should napari-micromanager just show *all* running mdas? regardless of how they were started?

tlambert03, 2024-01-26:
I edited the title here to say "programmatically"... note that if you hit the snap or live buttons:

[image]

or press the MDA button to open the MDA widget: 
[image]
and then run an MDA with 
[image]

it will populate the viewer with the images

linshaova, 2024-01-26:
Thank you! The suggested hack kind of works. I had to call the `fake_meta = ...` line every time before `run_mda(...)` though for the viewer to update with incoming images. If I `run_mda(...)` consecutively, the viewer's z slide moves, but the display remains black.

I vote for napari showing all MDAs regardless of the origin.

PR #308 (2024-01-28): Show output from any pymmcore-plus MDA in napari viewer
this makes it so that even if napari-micromanager didn't start the MDA, any MDA running in the currently monitored CMMCorePlus instance will be shown in the viewer.  Closes #305 

There are still places where bugs could potentially happen if frameReady somehow got emitted without sequenceStarted being emitted (not sure how that would happen), but those latent bugs have always been there
closed 2024-01-28 by tlambert03