pymmcore-plus/napari-micromanager #316: Help with napari micromanager

swmercer34, 2024-02-27:
The issue I am having is with saving an image to an allocated directory. As you can see in the screenshot, I set up the acquisition to save to the directory, however all I end up with is a weird test image with no data and also nothing is saved to the allocated file. Has anyone come across this issue before? Is there anything anyone can suggest? 
[image]

fdrgsp, 2024-02-27:
Hi @swmercer34, thanks for opening this issue.

Did you try to remove the spaces in the directory path (e.g. .`../Steve_test_data`)? I'm not sure what it can be the cause, I just run some test and I did not encounter any issue. 

Upgrading the  the data-saving functionality in `napari-micromanager` has been on our to-do-list and hopefully we will soon release an updated version where we use proper writers (we currently have an open PR specifically for that).

tlambert03, 2024-02-27:
Hi @swmercer34, thanks @fdrgsp.  

Yeah, we're currently struggling to reproduce this.  so we're going to need a bit more information.  I'll make this information easier to gather in an upcoming PR, but for now, could you run the following commands and paste the output here?

```
python --version
mmcore --version
pip list
```

thanks!

tlambert03, 2024-02-27:
Also, as @fdrgsp, the file-saving logic in napari-micromanager is lagging behind the activity in the lower level pymmcore-plus library... which now has much improved saving handlers for ome-tiff, ome-zarr, and tiff sequences.  We haven't yet plugged it into napari-micromanager though, but changes and improvements are coming soon