pymmcore-plus/napari-micromanager #328: Error logged when circular buffer is empty

rixius, 2024-03-14:
CoreViewerLink._update_viewer() causes MM to log lots of errors when the circular buffer is empty:

2024-03-14T11:12:27.488274 tid17768 [ERR,Core] Error occurred in device CMMCore::getLastImage: Circular buffer is empty.

Solution: I added

            if self._mmc.getRemainingImageCount() == 0:
                return

before the try-statement to suppress this.

(Note that the error is visible only when logging is enabled which I often do.)

tlambert03, 2024-03-14:
hey @rixius, I'm happy to add this.  And would be even happier if I could entice you to open a PR  :) it's nice to have others looking at the codebase.  Let me know if you don't have time and I can add it myself

rixius, 2024-03-14:
Sure, I will. I assumed that I require permissions first to push code to the repository and create a PR.

tlambert03, 2024-03-14:
> I assumed that I require permissions first to push code to the repository and create a PR.

You would make the changes to a personal fork, and then create a PR.  Happy to help get you started!  Here are the steps you would need to follow:

1. [Fork this repo](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/working-with-forks/fork-a-repo). On the main page of this repo click "fork"
    [image]
2. In your fork, [create a new branch](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-and-deleting-branches-within-your-repository)
3. In that new branch, make desired changes (you can either clone your fork to your local computer and then push them back to github, or you can [directly edit the file in github](https://docs.github.com/en/repositories/working-with-files/managing-files/editing-files)
4. Once you've made the changes [create a pull request](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork)

If any of that is confusing, or if you just want me to do it :) let me know!

rixius, 2024-03-14:
I think I'm getting there, need another try...

tlambert03, 2024-03-14:
no, no you did great!  you can open it again. if you're worried about the tests failing, don't worry, i think those were not your issue

tlambert03, 2024-03-14:
Thanks for the fix!

closed 2024-03-14 by tlambert03