pymmcore-plus/napari-micromanager #341: Unable to install mmcore

bluesupermn4, 2024-08-18:


bluesupermn4, 2024-08-18:
Trying to follow this install guide
https://www.napari-hub.org/plugins/napari-micromanager

I get to mmcore install and get the following error message

C:\WINDOWS\system32>mmcore install
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Program Files\Scripts\mmcore.exe\__main__.py", line 4, in <module>
  File "C:\Program Files\Lib\site-packages\pymmcore_plus\__init__.py", line 13, in <module>
    from .core import (
  File "C:\Program Files\Lib\site-packages\pymmcore_plus\core\__init__.py", line 23, in <module>
    from ._adapter import DeviceAdapter
  File "C:\Program Files\Lib\site-packages\pymmcore_plus\core\_adapter.py", line 5, in <module>
    from pymmcore_plus.core._constants import DeviceType
  File "C:\Program Files\Lib\site-packages\pymmcore_plus\core\_constants.py", line 6, in <module>
    import pymmcore
  File "C:\Program Files\Lib\site-packages\pymmcore\__init__.py", line 2, in <module>
    from .pymmcore_swig import *
  File "C:\Program Files\Lib\site-packages\pymmcore\pymmcore_swig.py", line 10, in <module>
    from ._pymmcore_swig import *
ImportError: DLL load failed while importing _pymmcore_swig: The specified module could not be found.

I am unable to move forward with the guide

Not sure what I am doing wrong

I have checked the requirements 
Requirements:

fonticon-materialdesignicons6
napari>=0.4.13
pymmcore-plus>=0.9.3
pymmcore-widgets>=0.7.0rc1
superqt>=0.5.1
tifffile
useq-schema>=0.4.1
zarr

All are satisfied

Any help would be greatly appreciated

tlambert03, 2024-08-18:
interesting.  looks like [pymmcore](https://github.com/micro-manager/pymmcore) itself isn't working for you (that's the python bindings for the micro-manager C++ layer, before you even get to the pymmcore-plus napari-micromanager stuff).

Can you try making a new environment, and only install pymmcore (`pip install pymmcore`), and then try just importing pymmcore?  `python -c "import pymmcore"`

if you get an error, please give me the full details for how you created an environment, (conda or other environment creation command, python version, etc...).  If you don't get an error, then continue on with `pip install pymmcore-plus`, and then `mmcore install`

bluesupermn4, 2024-08-19:
Hello Thanks for the reply. I tried again using the console in the napari gui instead of command line and was able to get the pluggin to show up in the gui now. However i am unable to get a live feed or snap using my bestscope camera, I directed the pluggin to the working config for micromanager and moved the DLL files for the camera into AppData\Local\napari-0.5.2\DLLs.

[image]

I am getting this error. Is there another place I should be putting the dll for the camera?

bluesupermn4, 2024-08-19:
i was able to figure it out. I just moved everything into this folder \AppData\Local\pymmcore-plus\pymmcore-plus\mm\Micro-Manager-2.0 manually and opened the config from there with the dll for the cameras in that folder as well

tlambert03, 2024-08-19:
Ok, If you've got something working great. If not, or in the future, I'll need more information on exactly how you've set everything up.

You don't need to put mm anywhere specific, but we do need to know where to look. mmcore install is a convenience to download them for you and put them in the default location, but there are many options:

see here for details: https://pymmcore-plus.github.io/pymmcore-plus/install/#installing-micro-manager-device-adapters

bluesupermn4, 2024-09-24:
I had further issues and ended up reinstalling via pip instead of trying to just use the GUI.exe installer

tlambert03, 2024-09-25:
Good idea :) 

No promises on the builtin napari installer, we don't control that

closed 2024-09-25 by tlambert03