pymmcore-plus/napari-micromanager #342: Unable to use my Prior II stage that works in micromanager

bluesupermn4, 2024-08-19:
Hello, I just started using the napari-micromanager plugin with my Olympus IX81 microscope that has a prior II control box for the stage and a BestScope camera. 

I took my working config from micromanager and opened it up in the plugin and found that everything seems to work as intended but the stage. When I try to open the stage controller I am met with an error. Any advice would be greatly appreciated

Also is there a way to save the menu configuration in this plugin so i dont have to open them and move them to where I want in the GUI everytime I start a new session?

fdrgsp, 2024-08-19:
Hi @bluesupermn4 , can you please provide the error message that you get? Otherwise it is hard to understand what is the issue. Thanks!

bluesupermn4, 2024-08-20:
[napari.txt](https://github.com/user-attachments/files/16677852/napari.txt)

bluesupermn4, 2024-08-20:
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
File ~\AppData\Local\napari-0.5.2\envs\napari-0.5.2\lib\site-packages\napari_micromanager\_gui_objects\_stages_widget.py:47, in MMStagesWidget._on_cfg_loaded(self=<napari_micromanager._gui_objects._stages_widget.MMStagesWidget(0x1cba70936e0)>)
     45     bx.setLayout(QHBoxLayout())
     46     bx.setSizePolicy(sizepolicy)
---> 47     bx.layout().addWidget(StageWidget(device=stage_dev))
        stage_dev = 'XYStage'
        bx = <napari_micromanager._gui_objects._stages_widget._DragGroupBox(0x1cba713ae90) at 0x000001CBAA4AAC80>
        StageWidget = <class 'pymmcore_widgets._stage_widget.StageWidget'>
     48     self.layout().addWidget(bx)
     49 self.resize(self.sizeHint())

File ~\AppData\Local\napari-0.5.2\envs\napari-0.5.2\lib\site-packages\pymmcore_widgets\_stage_widget.py:115, in StageWidget.__init__(self=<pymmcore_widgets._stage_widget.StageWidget(0x1cba713ab80)>, device='XYStage', levels=2, step=10, parent=None, mmcore=None)
    112 self._dtype = self._mmc.getDeviceType(self._device)
    113 assert self._dtype in STAGE_DEVICES, f"{self._dtype} not in {STAGE_DEVICES}"
--> 115 self._create_widget(step)
        self = <pymmcore_widgets._stage_widget.StageWidget(0x1cba713ab80) at 0x000001CBAA4AAB00>
        step = 10
    116 self._connect_events()
    117 self._set_as_default()

File ~\AppData\Local\napari-0.5.2\envs\napari-0.5.2\lib\site-packages\pymmcore_widgets\_stage_widget.py:157, in StageWidget._create_widget(self=<pymmcore_widgets._stage_widget.StageWidget(0x1cba713ab80)>, step=10)
    155 self._set_visible_levels(self._levels)  # type: ignore
    156 self._set_xy_visible()
--> 157 self._update_ttips()
        self = <pymmcore_widgets._stage_widget.StageWidget(0x1cba713ab80) at 0x000001CBAA4AAB00>
    159 self._readout = QLabel()
    160 self._readout.setAlignment(AlignCenter)

File ~\AppData\Local\napari-0.5.2\envs\napari-0.5.2\lib\site-packages\pymmcore_widgets\_stage_widget.py:323, in StageWidget._update_ttips(self=<pymmcore_widgets._stage_widget.StageWidget(0x1cba713ab80)>)
    321     continue
    322 if btn := item.widget():
--> 323     xmag, ymag = self.BTNS[f"{PREFIX}.{btn.text()}"][-2:]
        self = <pymmcore_widgets._stage_widget.StageWidget(0x1cba713ab80) at 0x000001CBAA4AAB00>
        self.BTNS = {'mdi6.\U000f0dbc': (0, 3, 0, 3), 'mdi6.\U000f013f': (1, 3, 0, 2), 'mdi6.\U000f0143': (2, 3, 0, 1), 'mdi6.\U000f0140': (4, 3, 0, -1), 'mdi6.\U000f013c': (5, 3, 0, -2), 'mdi6.\U000f0db9': (6, 3, 0, -3), 'mdi6.\U000f0dba': (3, 0, -3, 0), 'mdi6.\U000f013d': (3, 1, -2, 0), 'mdi6.\U000f0141': (3, 2, -1, 0), 'mdi6.\U000f0142': (3, 4, 1, 0), 'mdi6.\U000f013e': (3, 5, 2, 0), 'mdi6.\U000f0dbb': (3, 6, 3, 0)}
    324     if xmag:
    325         btn.setToolTip(f"move X by {self._scale(xmag)} µm")

KeyError: 'mdi6.?'

tlambert03, 2024-08-20:
that suggests that you didn't get a requirement installed properly...  Please respond to my request in https://github.com/pymmcore-plus/napari-micromanager/issues/341#issuecomment-2295374731

> if you get an error, please give me the full details for how you created an environment, (conda or other environment creation command, python version, etc...). If you don't get an error, then continue on with pip install pymmcore-plus, and then mmcore install

the issues you're running into are somewhat unusual, they suggest partial or strange installation of the requirements...  Can you tell us, from the moment of environment creation to the moment of starting napari-micromanager exactly what you did?

tlambert03, 2024-08-20:
alternatively, you can follow this exactly, and paste the full output of the windows command prompt (you can replace `mamba` with `conda` if you don't have mamba installed)

```
mamba create -n mm-env napari-micromanager pyqt
mamba activate napari-micromanager
mmcore install
mmcore info
napari -w napari-micromanager
```

then load your config at the top left and continue...

bluesupermn4, 2024-09-24:
The issue was resolved once I installed everything via pip instead of trying to go through the GUI.exe

tlambert03, 2024-09-25:
Great, glad to hear it.  Yep that's the recommended approach

closed 2024-09-25 by tlambert03