pymmcore-plus/napari-micromanager #343: Shutter not re-opening when channel is changed during live mode

simonecoppola, 2024-08-21:
We are having some issue with changing channels when in Live Mode. Essentially, if the channel (which LED is on) is changed during the live mode, the LED gets turned off, but the new LED does not get turned on. The change only fully occurs when the live mode is stopped and restarted it.

I'm not sure if that's the best way, but I resolved this bug by adding this connection to the `_core_link.py` code:
`self._mmc.events.configSet, self._restart_live`

tlambert03, 2024-08-23:
thanks for the very specific help @simonecoppola, i'm find with that fix.  would you be willing to submit a PR?  (if not, no worries i can submit it)

simonecoppola, 2024-08-23:
Great! I've just submitted the PR (I'm not super familiar with that aspect of Github yet so I hope I've done it right!)

And thank you for your work on pymmcore-plus, it's great! I have a lot more questions but will open other issues as I go :)

tlambert03, 2024-08-24:
Thanks for the PR!  

Glad you're finding pymmcore-plus useful.  Please feel free to ask away :)  
If you'd like more real-time discussion, I also pay attention on the Imagesc zulip chat, and started a channel there: https://imagesc.zulipchat.com/#narrow/stream/442785-pymmcore.5B-plus.5D

closed 2024-08-24 by tlambert03