pymmcore-plus/napari-micromanager #346: Trying to use my BestScope camera for MDA

bluesupermn4, 2024-09-24:
I am attempting to use my BestScope BUC5F-830EC camera with the Napari-micromanager plugin's MDA Z stack. It works fine for me in micromanager but I am getting this error in Napari 

Any help would be greatly appreciated

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
File c:\users\lab\appdata\local\programs\python\python39\lib\site-packages\superqt\utils\_qthreading.py:613, in create_worker.<locals>.reraise(e=ValueError("parameter 'value': expected array with shape (2160, 3840), got (2160, 3840, 3)"))
    612 def reraise(e):
--> 613     raise e
        e = ValueError("parameter 'value': expected array with shape (2160, 3840), got (2160, 3840, 3)")

File c:\users\lab\appdata\local\programs\python\python39\lib\site-packages\superqt\utils\_qthreading.py:175, in WorkerBase.run(self=<superqt.utils._qthreading.GeneratorWorker object>)
    173     warnings.filterwarnings("always")
    174     warnings.showwarning = lambda *w: self.warned.emit(w)
--> 175     result = self.work()
        self = <superqt.utils._qthreading.GeneratorWorker object at 0x000002159D4B0430>
    176 if isinstance(result, Exception):
    177     if isinstance(result, RuntimeError):
    178         # The Worker object has likely been deleted.
    179         # A deleted wrapped C/C++ object may result in a runtime
    180         # error that will cause segfault if we try to do much other
    181         # than simply notify the user.

File c:\users\lab\appdata\local\programs\python\python39\lib\site-packages\superqt\utils\_qthreading.py:440, in GeneratorWorker.work(self=<superqt.utils._qthreading.GeneratorWorker object>)
    438 try:
    439     _input = self._next_value()
--> 440     output = self._gen.send(_input)
        self = <superqt.utils._qthreading.GeneratorWorker object at 0x000002159D4B0430>
        _input = None
        self._gen = <generator object _NapariMDAHandler._watch_mda at 0x000002159C8CF890>
    441     self.yielded.emit(output)
    442 except StopIteration as exc:

File c:\users\lab\appdata\local\programs\python\python39\lib\site-packages\napari_micromanager\_mda_handler.py:141, in _NapariMDAHandler._watch_mda(self=<napari_micromanager._mda_handler._NapariMDAHandler object>)
    139 while self._mda_running:
    140     if self._deck:
--> 141         layer_name, im_idx = self._process_frame(*self._deck.pop())
        self = <napari_micromanager._mda_handler._NapariMDAHandler object at 0x00000215E4611C40>
        self._deck = deque([(array([[[255, 255, 255],
        [255, 255, 255],
        ...,
        [255, 255, 255],
        [255, 255, 255]],

       [[255, 255, 255],
        [255, 255, 255],
        ...,
        [255, 255, 255],
        [255, 255, 255]],

       ...,

       [[255, 255, 255],
        [255, 255, 255],
        ...,
        [255, 255, 255],
        [255, 255, 255]],

       [[255, 255, 255],
        [255, 255, 255],
        ...,
        [255, 255, 255],
        [255, 255, 255]]], dtype=uint8), MDAEvent(index=mappingproxy({'p': 0, 'c': 0, 'z': 1}), channel=Channel(config=''), exposure=100.0, x_pos=-32305.709, y_pos=-0.038, z_pos=441.631)), (array([[[255, 255, 255],
        [255, 255, 255],
        ...,
        [255, 255, 255],
        [255, 255, 255]],

       [[255, 255, 255],
        [255, 255, 255],
[... 313 lines cut ...]
        [255, 255, 255]],

       [[255, 255, 255],
        [255, 255, 255],
        ...,
        [255, 255, 255],
        [255, 255, 255]],

       ...,

       [[255, 255, 255],
        [255, 255, 255],
        ...,
        [255, 255, 255],
        [255, 255, 255]],

       [[255, 255, 255],
        [255, 255, 255],
        ...,
        [255, 255, 255],
        [255, 255, 255]]], dtype=uint8)
        param = 'value'
    565                      .format(param, shape, array.shape))

ValueError: parameter 'value': expected array with shape (2160, 3840), got (2160, 3840, 3)

tlambert03, 2024-09-26:
thanks @bluesupermn4, yes it looks like napari-micromanager wasn't expecting color cameras and RGB images.  

@fdrgsp, we need to fix that here:
https://github.com/pymmcore-plus/napari-micromanager/blob/31d973d2045505bd8c8d88e38f153da5c39d153f/src/napari_micromanager/_mda_handler.py#L96-L108

tlambert03, 2024-09-26:
fix incoming in #347

closed 2024-09-26 by tlambert03

tlambert03, 2024-09-26:
@bluesupermn4 , please update with `pip install -U napari-micromanager` and try again

bluesupermn4, 2024-09-30:
That did it! Thank you Thank you

PR #347 (2024-09-26): fix: fix rgb MDA
fixes #346