pymmcore-plus/napari-micromanager #353: Issue with running napari-micromanager and mda engine in the same Jupyter notebook

alandolt, 2024-12-19:
Hello,

First, I would like to express my gratitude for the development of pymmcore-plus, the useq schemes, pymmcore-widgets, and napari-micromanager—they are fantastic tools.

I am currently working on integrating napari-micromanager with an MDA acquisition running in a Jupyter notebook. The goal is to use napari-micromanager (or later, specific pymmcore-widgets) to set up experiments, select FOVs, and determine the correct exposure time. This works well with the data_mda widget, which allows us to select FOVs. We can then read these FOVs in the Jupyter notebook and use the pymmcore MDA engine to incorporate them into an experiment.

However, we are encountering an issue where napari-micromanager interferes with run_mda when both are launched in the same Jupyter notebook. Specifically, frames are duplicated (they appear in both napari-micromanager and as two frameReady signals), and we receive the following error at the end of an MDA acquisition: (started with the pymmcore mda engine)

```
2024-12-19T18:43:14.062    tid0x434c [ERROR,pymmcore-plus] 

While emitting signal 'pymmcore_plus.mda.events._psygnal.MDASignaler.sequenceFinished', a AttributeError occurred in a callback:

  Signal emitted at: C:\Users\Niesen\Documents\alandolt\code\pymmcore-plus\src\pymmcore_plus\mda\_runner.py:417, in _finish_run
    >  self._signals.sequenceFinished.emit(sequence)

  Callback error at: c:\Users\Niesen\.conda\envs\pymmcore\Lib\site-packages\napari_micromanager\_mda_handler.py:335, in _id_idx_layer
    >  meta = cast(dict, seq.metadata.get(NMM_METADATA_KEY, {}))

    Local variables:
       event = MDAEvent(index={'t': 2}, channel=Channel(config='CyanStim', ...
       seq = None

See AttributeError above for original traceback.
```

I have tried creating a separate MDA engine with the following code:

```
from pymmcore_plus.mda import MDAEngine
new_mda_engine = MDAEngine(mmc)
mmc.mda.set_engine(new_mda_engine)
mmc.register_mda_engine(new_mda_engine)
```

However, this has not resolved the issue.

Here is a minimal code example that demonstrates the issue:

**Cell to start pymmcore**
```
mmc = pymmcore_plus.CMMCorePlus()
mmc.loadSystemConfiguration("E:\\MicroManagerConfigs\\Ti2CicercoConfig_w_DMD_9_w_ttl.cfg")
```

**Cell for loading napari-micromanager**
```
viewer = napari.Viewer()
viewer.window.add_plugin_dock_widget("napari-micromanager")
```

**Cell to then start the experiment**
```
q = Queue()                    # create the queue
STOP = object()                # any object can serve as the sentinel
q_iterator = iter(q.get, STOP) # create the queue-backed iterable

# start the acquisition in a separate thread
mmc.run_mda(q_iterator)

# (optional) connect some callback to the imageReady signal
@mmc.mda.events.frameReady.connect
def on_frame(img, event):
    print(f'Frame {event.index} received: {img.shape}')

ev1 = useq.MDAEvent(index= {"t": 0}, channel={"config": "CyanStim", "group": "CF_DMD"}, exposure=1000, min_start_time=0, reset_event_timer=True)
ev2 = useq.MDAEvent(index= {"t": 1}, channel={"config": "CyanStim", "group": "CF_DMD"}, exposure=1000, min_start_time=10.0)
ev3 = useq.MDAEvent(index= {"t": 2}, channel={"config": "CyanStim", "group": "CF_DMD"}, exposure=1000, min_start_time=15.0)

q.put(ev1)
q.put(ev2)
q.put(ev3)
q.put(STOP)
```

**Output of mmcore info:** 
```
python              : 3.11.10 | packaged by conda-forge | (main, Oct 16 2024, 01:17:14) [MSC v.1941 64 bit (AMD64)]
platform            : Windows-10-10.0.22631-SP0
pymmcore-plus       : 0.12.1.dev42+g7f19a67
pymmcore            : 11.1.1.71.3
core-version-info   : MMCore version 11.1.1
api-version-info    : Device API version 71, Module API version 10
adapter-path        : C:\Program Files\Micro-Manager-2.0
useq-schema         : 0.6.3.dev1+g0199fa3
pymmcore-widgets    : 0.8.0
qt                  : PyQt5 5.15.8
napari-micromanager : 0.1.3
napari              : 0.5.4
tifffile            : 2024.9.20
zarr                : 2.18.3
```

Has anyone encountered a similar issue or have any suggestions for how to resolve this conflict between napari-micromanager and MDA acquisitions in a Jupyter notebook?

Any assistance would be greatly appreciated. Thank you in advance!

tlambert03, 2024-12-19:
thanks for opening the issue and for all the info!  Glad it's mostly been helpful for you and let's definitely get this fixed.

one more thing that would be useful here is to see the error *above* where it says `See AttributeError above for original traceback.`. so, if you can, scroll up to where it says something like "was the direct cause of the next exception" and show my what that attribute error is.

I"ll also try to replicate this locally when I have a moment.  I haven't tried to use both a notebook and napari like this before, but in theory it should definitely be fine.

alandolt, 2024-12-19:
Thank you very much for your fast reply. 
Yes, it's really amazing, together with hinderling, I'm trying to set up a DMD stimulation live feedback pipeline, where pymmcore and the associated pymmcore-widgets are perfect for providing a nice GUI and the microscope-python interface.  

Sorry for not having provided the full error log, here are the missing parts: 
```
2024-12-19T18:46:03.674282 tid6156 [IFO,Core] Default camera set to HamamatsuHam_DCAM
2024-12-19T18:46:03.670    tid0x180c [INFO,pymmcore-plus] MDA Started: GeneratorMDASequence()
2024-12-19T18:46:03.881    tid0x180c [INFO,pymmcore-plus] index={'t': 0} channel=Channel(config='CyanStim', group='CF_DMD') exposure=1000.0 min_start_time=0.0 reset_event_timer=True
2024-12-19T18:46:13.880    tid0x180c [INFO,pymmcore-plus] index={'t': 1} channel=Channel(config='CyanStim', group='CF_DMD') exposure=1000.0 min_start_time=10.0
2024-12-19T18:46:18.880    tid0x180c [INFO,pymmcore-plus] index={'t': 2} channel=Channel(config='CyanStim', group='CF_DMD') exposure=1000.0 min_start_time=15.0
2024-12-19T18:46:20.004    tid0x180c [INFO,pymmcore-plus] MDA Finished: GeneratorMDASequence()
2024-12-19T18:46:20.013    tid0x180c [ERROR,pymmcore-plus] 

While emitting signal 'pymmcore_plus.mda.events._psygnal.MDASignaler.sequenceFinished', a AttributeError occurred in a callback:

  Signal emitted at: C:\Users\Niesen\Documents\alandolt\code\pymmcore-plus\src\pymmcore_plus\mda\_runner.py:417, in _finish_run
    >  self._signals.sequenceFinished.emit(sequence)

  Callback error at: c:\Users\Niesen\.conda\envs\pymmcore\Lib\site-packages\napari_micromanager\_mda_handler.py:335, in _id_idx_layer
    >  meta = cast(dict, seq.metadata.get(NMM_METADATA_KEY, {}))

    Local variables:
       event = MDAEvent(index={'t': 2}, channel=Channel(config='CyanStim', ...
       seq = None

See AttributeError above for original traceback.
```

tlambert03, 2024-12-19:
ahhh, ok I think i see what's happening here.  and I *think* it has more to do with using Queue than it does with using jupyter.

if you just run a basic timelapse with, say:

```python
mmc.run_mda(useq.MDASequence(time_plan=useq.TIntervalLoops(loops=5, interval=1)))
```

does it work fine?

The issue here is unfortunately slightly more complex than a simple fix.  `napari-micromanager` is currently not really set up for the non-deterministic Queue style of acquisition.  It pretty much expects a "classical" array with known dimensions, and more specifically, the Attribute error here is coming when it looks for the `MDASequence` object on the `MDAEvent` (which, of course, is not present when the MDAEvent has been created manually as has been done here).

Your code and use case is perfectly valid, we just need to update the code here to be a bit smarter.  A fair amount of work has been done at lower levels (e.g. with the [TensorStoreHandler](https://github.com/pymmcore-plus/pymmcore-plus/blob/main/src/pymmcore_plus/mda/handlers/_tensorstore_handler.py)) to handle this exact case and fallback gracefully to a simple time series of images (without higher dimensionality awareness), and napari-micromanager should be capable of showing you that.  but it will require a bit of rearranging.

pinging @fdrgsp, in case we can possible nerd snipe him and his time here :)

alandolt, 2024-12-19:
Thank you very much for your answer. Yes the basic example works. 
Also thank you very much for the detailed explanation on the underlying issue. 
Is there maybe an easy way to instead simple disconnect napari-micromanager from the mda acquisition engine (sorry I didn't really found anything regarding this in the documenation)? I ask as data storage is anyway handled by my script and I even don't need napari to display the images. 
Napari-micromanager would only be used to initially set the correct exposure parameters, as well as with the well plate tool to automatically generate fovs from wells of my well plate.

tlambert03, 2024-12-21:
> Is there maybe an easy way to instead simple disconnect napari-micromanager from the mda acquisition engine

yes, that would be possible!   You need to get handle on the actual micro-manager widget.  one way to do this would be to instantiate it directly, rather than using `add_plugin_dock_widget`:

```python
from napari_micromanager import MainWindow

viewer = napari.Viewer()
mm_wdg = MainWindow(viewer)
viewer.window.add_dock_widget(mm_wdg)
```

and then to disconnect the viewer from all the image showing stuff, I think you should just be able to:

```python
mm_wdg._core_link.cleanup()
```

----------

more generally, if you really didn't need napari for anything other than the widgets.  We'd be more than happy to help you create a custom set of pymmcore-widgets that have everything you want without all the additional weight of napari.  And there are simpler (less fully featured) ways to view the images as well that would work for this application.

I can dig into that in a couple weeks after I'm back from winter break.

alandolt, 2025-01-16:
Thank you so much for your offer and your valuable help!

Currently, the project is working well using the code you provided. For the sake of completeness—especially for others who may be looking to integrate napari-micromanager with Jupyter notebooks—if you'd like to reconnect napari-micromanager with the core, you can use the following code:
```
from napari_micromanager._core_link import CoreViewerLink
mm_wdg._core_link = CoreViewerLink(viewer, mmc)
```

There is still a warning that appears whenever executing code for an acquisition from the notebook while disconnected from napari-micromanager. However, I believe it can be safely ignored:
 
```
WARNING: QBasicTimer::start: QBasicTimer can only be used with threads started with QThread
```

I would be happy to take you up on your offer once we make more progress on the real-time feedback microscopy pipeline based on pymmcore. In that pipeline, we are extensively using the MDA Widget, PropertiesWidget, and the live view prior to starting the acquisition process. As soon as we have a working initial version, I will contact you again.

tlambert03, 2025-01-16:
thanks for the update.  Yes feel free to ping.  lots of stuff in the works.  We'll have more direct support for https://github.com/pyapp-kit/ndv in the near future, which will make usage in jupyter even easier (keeping napari out of the loop entirely)

closed 2025-01-16 by tlambert03